RobotLocomotion/drake #2329: Simplify cmake download-all target

billhoffman, 2016-05-12:
Use ExternalProject_Add_StepTargets to create the download-all target so that you do not have to break the External projects into a project and a project-download.  This will fix a race condition with ninja.

bradking, 2016-05-12:
The `ExternalProject_Add_StepTargets` feature has been around since CMake 2.8.3, but we need the `NO_DEPENDS` option which has only existed since CMake 3.2.

david-german-tri, 2016-06-08:
#2487 will close this.

david-german-tri, 2016-06-22:
From weekly sync: can we just remove the `download-all` target?

bradking, 2016-06-22:
For reference, this issue as originally recorded was addressed by #2487 and can therefore be closed.

> can we just remove the download-all target?

Perhaps, but IMO we should discuss it in a separate issue to avoid confusion.

closed 2016-06-22 by bradking