RobotLocomotion/drake #2330: 'lcmtypes/bot_core/planar_lidar_t.hpp' file not found

jamiesnape, 2016-05-12:
Transient error on OS X builds:
```
/Users/yosemite/workspace/mac-clang-continuous-matlab-open-source/drake/systems/plants/test/lidarTest.cpp:3:10: fatal error: 'lcmtypes/bot_core/planar_lidar_t.hpp' file not found
#include "lcmtypes/bot_core/planar_lidar_t.hpp"
         ^
1 error generated.
```

RussTedrake, 2016-05-12:
I have seen this before if the dependencies are not quite correct on parallel builds.  The auto-generated file doesn’t get built before the code that depends on it...

liangfok, 2016-05-12:
I believe I know the fix for this and will issue a PR shortly.

jamiesnape, 2016-05-16:
Same problem in the latest continuous build:
https://drake-jenkins.csail.mit.edu/job/mac-clang-continuous-debug/168/consoleText
https://drake-cdash.csail.mit.edu/viewBuildError.php?onlydeltap&buildid=20700

liangfok, 2016-05-21:
Looking at the [mac-clang-continuous-debug](https://drake-jenkins.csail.mit.edu/job/mac-clang-continuous-debug/168/consoleText) log file, here are the places that mention "planar_lidar_t":

On line 12,955, Java code is generated:
```
12955 [ 10%] Generating lcmgen/lcmtypes/bot_core/planar_lidar_t.java
12956 cd /Users/yosemite/workspace/mac-clang-continuous-debug/pod-build/externals/bot_core_lcmtypes/lcmgen/lcmtypes && /Users/yosemite/workspace/mac-clang-continuous-debu      g/build/bin/lcm-gen --jdefaultpkg lcmtypes --java /Users/yosemite/workspace/mac-clang-continuous-debug/externals/bot_core_lcmtypes/lcmtypes/bot_core_planar_lidar_t.      lcm
```

On line 38,574, `lidarTest.cpp` fails to include `planar_lidar_t.hpp`:
```
38574 /Users/yosemite/workspace/mac-clang-continuous-debug/drake/systems/plants/test/lidarTest.cpp:8:10: fatal error: 'lcmtypes/bot_core/planar_lidar_t.hpp' file not foun      d
38575 #include "lcmtypes/bot_core/planar_lidar_t.hpp"
38576          ^
38577 1 error generated.
```

On line 46,073, `lidarTest.cpp` again fails to include `planar_lidar_t.hpp`:
```
46073 Preprocessing /Users/yosemite/workspace/mac-clang-continuous-debug/drake/systems/plants/test/testRigidBodyInertialMeasurementUnit.m.../Users/yosemite/workspace/mac-      clang-continuous-debug/drake/systems/plants/test/lidarTest.cpp:8:10: fatal error: 'lcmtypes/bot_core/planar_lidar_t.hpp' file not found
46074 #include "lcmtypes/bot_core/planar_lidar_t.hpp"
46075          ^
46076 1 error generated.
```
It's strange how only the Java code is generated. I wonder whether there are any print statements indicating that the C++ code is generated.

liangfok, 2016-05-22:
Since this issue is a duplicate of https://github.com/RobotLocomotion/drake/issues/2341, I'm going to close this issue.

closed 2016-05-22 by liangfok