RobotLocomotion/drake #2333: MATLAB tests passing despite "error in addpath_drake"

jamiesnape, 2016-05-12:
Illustrative example. Why there was the error is not the subject of this ticket. The main point is that the test still passed.
```
418: Start 418: examples/ZMP/test/testZMPplan_coordinates
418: Test command: /usr/local/Cellar/cmake/3.5.2/bin/cmake "-Dtest_name=examples/ZMP/test/testZMPplan_coordinates" "-Dadditional_paths=/Users/yosemite/workspace/mac-clang-experimental-matlab-open-source/build/matlab;/Users/yosemite/workspace/mac-clang-experimental-matlab-open-source/drake/examples/ZMP/test" "-Dtest_timeout=-1" "-Doutput_directory=/Users/yosemite/workspace/mac-clang-experimental-matlab-open-source/drake/pod-build/Matlab" "-Dworking_directory=/Users/yosemite/workspace/mac-clang-experimental-matlab-open-source/drake/examples/ZMP/test" "-DMatlab_PROGRAM=/Applications/MATLAB_R2015b.app/bin/matlab" "-Dno_unittest_framework=FALSE" "-DMatlab_ADDITIONNAL_STARTUP_OPTIONS=" "-Dunittest_file_to_run=" "-Dcustom_Matlab_test_command=try, eval('testZMPplan_coordinates'); catch ex, disp(getReport(ex,'extended')); disp(' '); force_close_system; exit(~strncmp(ex.identifier,'Drake:MissingDependency',23)); end; force_close_system; exit(0)" "-Dcmd_to_run_before_test=addpath_drake; global g_disable_visualizers; g_disable_visualizers=true" "-P" "/Users/yosemite/workspace/mac-clang-experimental-matlab-open-source/drake/cmake/MatlabTestsRedirect.cmake"
418: {Error using addpath_drake (line 24)
418: You must run make first (and/or add your pod build/matlab directory to the
418: matlab path)
418: 
418: Error in addpath_drake (line 4)
418:   addpath_drake();
418: } 
418: Matlab test  output:
418: 
418:                             < M A T L A B (R) >
418:                   Copyright 1984-2015 The MathWorks, Inc.
418:                    R2015b (8.6.0.267246) 64-bit (maci64)
418:                               August 20, 2015
418: 
418:  
418: To get started, type one of these: helpwin, helpdesk, or demo.
418: For product information, visit www.mathworks.com.
418:  
418: 
418:     Academic License
418: 
418: Adding /Users/yosemite/workspace/mac-clang-experimental-matlab-open-source/drake/../build/matlab to the matlab path
418: {Error using addpath_drake (line 24)
418: You must run make first (and/or add your pod build/matlab directory to the
418: matlab path)
418: 
418: Error in addpath_drake (line 4)
418:   addpath_drake();
418: } 
418: 
418: Test #418: examples/ZMP/test/testZMPplan_coordinates ......................................   Passed   10.80 sec
```
FYI @RussTedrake

billhoffman, 2016-05-13:
Here is a link to a dashboard build: https://drake-cdash.csail.mit.edu/testDetails.php?test=640036&build=18882

RussTedrake, 2016-05-25:
related to #2416.  something fishy happening here.

david-german-tri, 2016-06-08:
At a high level, MATLAB is returning 0 in this case, and we'd need it to return 1.  Specifically, we need the following command to fail:

`https://github.com/RobotLocomotion/drake/blob/0294fdbee04d18ab8421c8de759d2860ca3f4f7a/drake/CMakeLists.txt#L189`

david-german-tri, 2017-04-13:
Closing as wontfix.  This would be challenging to fix; since the legacy MATLAB code is no longer a priority, let's not bother.

closed 2017-04-13 by david-german-tri