RobotLocomotion/drake #2338: 'drake/drakeRBSystem_export.h' file not found

jamiesnape, 2016-05-13:
```
In file included from /Users/yosemite/workspace/mac-clang-continuous-matlab/drake/systems/plants/RigidBodyTree.cpp:1:
In file included from /Users/yosemite/workspace/mac-clang-continuous-matlab/drake/../drake/systems/plants/RigidBodyTree.h:15:
/Users/yosemite/workspace/mac-clang-continuous-matlab/drake/../drake/systems/plants/RigidBodyFrame.h:5:10: fatal error: 'drake/drakeRBSystem_export.h' file not found
#include "drake/drakeRBSystem_export.h"
         ^
1 error generated.
```

```
In file included from /Users/yosemite/workspace/mac-clang-continuous-matlab/drake/systems/plants/RigidBodyTreeURDF.cpp:7:
In file included from /Users/yosemite/workspace/mac-clang-continuous-matlab/drake/../drake/systems/plants/rigid_body_tree_urdf.h:7:
/Users/yosemite/workspace/mac-clang-continuous-matlab/drake/../drake/systems/plants/RigidBodyFrame.h:5:10: fatal error: 'drake/drakeRBSystem_export.h' file not found
#include "drake/drakeRBSystem_export.h"
         ^
1 error generated.
```

https://drake-cdash.csail.mit.edu/viewBuildError.php?onlydeltap&buildid=19459

avalenzu, 2016-05-14:
I'm seeing this on an OSX Yosemite machine as well.

avalenzu, 2016-05-15:
To follow up, on my machine this was caused by #2357

david-german-tri, 2016-05-25:
#2357 resolves this opaque build failure, but does not address why LCM is sometimes absent.

liangfok, 2016-05-25:
I believe this is a duplicate of https://github.com/RobotLocomotion/drake/issues/2341.

david-german-tri, 2016-05-25:
Great, thanks!  Closing this one.

closed 2016-05-25 by david-german-tri