RobotLocomotion/drake #2339: systems/plants/constraint/test/testKinCnst hangs on Windows 32-bit builds

jamiesnape, 2016-05-13:
```
122: Test command: "C:\Program Files (x86)\CMake\bin\cmake.exe" "-Dtest_name=systems/plants/constraint/test/testKinCnst" "-Dadditional_paths=C:/projects/ReYFE7LH/build/matlab;C:/projects/ReYFE7LH/drake/systems/plants/constraint/test" "-Dtest_timeout=-1" "-Doutput_directory=C:/projects/ReYFE7LH/drake/pod-build/Matlab" "-Dworking_directory=C:/projects/ReYFE7LH/drake/systems/plants/constraint/test" "-DMatlab_PROGRAM=C:/Program Files (x86)/MATLAB/R2015b/bin/matlab.exe" "-Dno_unittest_framework=FALSE" "-DMatlab_ADDITIONNAL_STARTUP_OPTIONS=" "-Dunittest_file_to_run=" "-Dcustom_Matlab_test_command=try, eval('testKinCnst'); catch ex, disp(getReport(ex,'extended')); disp(' '); force_close_system; exit(~strncmp(ex.identifier,'Drake:MissingDependency',23)); end; force_close_system; exit(0)" "-Dcmd_to_run_before_test=addpath_drake; global g_disable_visualizers; g_disable_visualizers=true" "-P" "C:/projects/ReYFE7LH/drake/cmake/MatlabTestsRedirect.cmake"
Test #122: systems/plants/constraint/test/testKinCnst .....................................***Timeout 3000.00 sec
```
(no other console output)

RussTedrake, 2016-05-25:
+ @sammy-tri

hongkai-dai, 2016-05-25:
I do not  have a Windows 32 bit machine to test this. Do we have a Windows machine in the lab?

jamiesnape, 2016-10-05:
Closing since Windows is no longer supported.

closed 2016-10-05 by jamiesnape