RobotLocomotion/drake #2341: lcm-java: 'c' flag requires manifest or input files to be specified

jamiesnape, 2016-05-13:
Transcient issue with OS X builds.

**Command**: `"/Library/Java/JavaVirtualMachines/jdk1.8.0_92.jdk/Contents/Home/bin/jar" "-cf" "/Users/yosemite/workspace/mac-clang-continuous-matlab/pod-build/externals/lcm/lcm-prefix/src/lcm-build/lcm-java/lcm.jar" "@java_class_filelist"`

**Directory**: `/Users/yosemite/workspace/mac-clang-continuous-matlab/pod-build/externals/lcm/lcm-prefix/src/lcm-build/lcm-java/CMakeFiles/lcm-java.dir`

**Output**:
```
'c' flag requires manifest or input files to be specified!
Usage: jar {ctxui}[vfmn0PMe] [jar-file] [manifest-file] [entry-point] [-C dir] files ...
Options:
    -c  create new archive
    -t  list table of contents for archive
    -x  extract named (or all) files from archive
    -u  update existing archive
    -v  generate verbose output on standard output
    -f  specify archive file name
    -m  include manifest information from specified manifest file
    -n  perform Pack200 normalization after creating a new archive
    -e  specify application entry point for stand-alone application 
        bundled into an executable jar file
    -0  store only; use no ZIP compression
    -P  preserve leading '/' (absolute path) and ".." (parent directory) components from file names
    -M  do not create a manifest file for the entries
    -i  generate index information for the specified jar files
    -C  change to the specified directory and include the following file
If any file is a directory then it is processed recursively.
The manifest file name, the archive file name and the entry point name are
specified in the same order as the 'm', 'f' and 'e' flags.

https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=19421

jamiesnape, 2016-05-17:
Also https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=21832.

david-german-tri, 2016-05-18:
Happened again here: https://drake-jenkins.csail.mit.edu/job/experimental/1534/compiler=clang,label=mac/consoleFull

liangfok, 2016-05-22:
I believe the relevant code is in `[drake-distro]/externals/lcm/lcm/lcm-src/lcm-java/CMakeLists.txt` line 28.

```
add_jar(lcm-java
  SOURCES ${lcm_java_files} ${jchart2d_files}
  OUTPUT_NAME lcm)
```

Assuming the above is correct, I think we should modify the file to log the values of `${lcm_java_files}` and `${jchart2d_files}` so we know exactly what's being passed to `jar`.

david-german-tri, 2016-06-01:
Blocked by #2486.

PR #2582 (2016-06-28): Port to new CMake-ified LCM, bot_core_lcmtypes
Replace use of `lcmtypes.cmake` with functions provided by upstream LCM Use file. Fix compilation of JAR (now requires specifying dependency JAR's rather than relying on modification of the global `CLASSPATH`). Remove references to `lcmtype_agg_hpp` target; the aggregate headers are created at configure time, and the per-type headers are generated at the same time as the C sources for the lcmtypes library (which, being itself a dependency of consumers of the C++ headers, should suffice to ensure that said headers exist before trying to build sourc [... cut]
closed 2016-06-28 by david-german-tri