RobotLocomotion/drake #2344: How should developers build documentation?

david-german-tri, 2016-05-13:
The successor to `make doc` should be documented on the wiki.

jwnimmer-tri, 2016-05-13:
Also, throw in retroactive `CHANGELOG.md` for #2143 while we're at it?

jamiesnape, 2016-05-13:
I disagree with the CHANGELOG issue. For development, the git history is the CHANGELOG. Someone should generate the changelogs (semi-)automatically at release time.

david-german-tri, 2016-05-13:
@jamiesnape I think that's the long-term goal, but right now we have dependent projects and we do not have stable resources, so we need to communicate breaking API changes to them in some way.

That said, I don't think #2143 requires a CHANGELOG update because it's not reeeeally an API change.

jwnimmer-tri, 2016-05-13:
Okay, I retract my suggestion.

jamiesnape, 2016-05-13:
Aside: Do we have a release process yet?

david-german-tri, 2016-05-13:
> Do we have a release process yet?

Nope!  I think there's little point in having releases until we have a defined API, and that's a ways off.

jamiesnape, 2016-05-13:
That is very true.

bradking, 2016-05-13:
> the git history is the CHANGELOG. Someone should generate the CHANGELOG (semi-)automatically at release time.

Git history is too busy to call out user-facing changes that would normally be maintained in a CHANGELOG or other form of release notes.  However, the structure of a CHANGELOG is not well-suited to being maintained in DVCS because appending notes to it will always conflict.  What I've done in other projects is have a release note staging directory in which release notes can be added with some arbitrary file name that is different for each topic.  That will never conflict.  Then the release manager consolidates the notes into a well-organized file just before the release, and associates them with the release version number in which they appear.  This is also a chance to polish them.

closed 2016-05-17 by sherm1