RobotLocomotion/drake #2346: System: Design how to use `FunctionalForm` to extract a system form

bradking, 2016-05-13:
Discussion of how to do this (for #1666) started in #2322 ([here](https://github.com/RobotLocomotion/drake/pull/2322#issuecomment-219008200), [here](https://github.com/RobotLocomotion/drake/pull/2322#issuecomment-219057261), and [here](https://github.com/RobotLocomotion/drake/pull/2322#issuecomment-219120729)) but is tangential to that PR and so can move to this separate issue.

bradking, 2016-05-13:
Cc: @david-german-tri @RussTedrake

bradking, 2016-05-13:
> That is super cool - using FunctionalForm as an Eigen scalar will be a great way to getting insight into complex systems, and the code you pasted above has a straightforward System2 analogue.

Good.

> I don't think it's a complete answer for how to label Systems, though, because it only gives you an answer for a particular code path, which could change in response to changing simulation state.

Yes, for System or Function concept implementations that might change their state during a simulation or an optimization then custom form logic may be needed.  The default assumption is that everything changing during an optimization should be captured in one of the decision variables.  Maybe this is not true for simulations, but I'm not familiar enough with that infrastructure yet.

> So, it seems likely the programmer will have to bear some of the burden of specifying (conservatively) the form/sparsity matrix.

The approach I've had in mind for this is to optionally provide an overload of `eval()/output()/dynamics()` that specializes the implementation for `ScalarType == FunctionalForm` and computes form information.  This will be necessary at least when the main overload of one of these methods does operations that cannot be captured by `FunctionalForm` operators, such as conditions (`if(a < b)`).

david-german-tri, 2017-01-13:
The notes here are largely superseded by System2, where we plan to so something analogous to `ToAutoDiffXd` for symbolic/FunctionalForm.

closed 2017-01-13 by david-german-tri