RobotLocomotion/drake #2347: Unrecognized file extension in compareParsers test

jamiesnape, 2016-05-13:
```
testing /home/ubuntu/workspace/linux-gcc-continuous-matlab/drake/examples/UnderwaterAcrobot/UnderwaterAcrobot.urdf
 this model has added mass (not implemented in c++ yet)
testing /home/ubuntu/workspace/linux-gcc-continuous-matlab/drake/examples/Pendulum/Pendulum.urdf
testing /home/ubuntu/workspace/linux-gcc-continuous-matlab/drake/examples/Pendulum/test/p10.urdf
testing /home/ubuntu/workspace/linux-gcc-continuous-matlab/drake/examples/Pendulum/test/p6.urdf
testing /home/ubuntu/workspace/linux-gcc-continuous-matlab/drake/examples/Pendulum/test/p7.urdf
testing /home/ubuntu/workspace/linux-gcc-continuous-ma
Error using RigidBodyManipulator (line 79)
Unrecognized file extension:  for file linux-gcc-continuous-ma

Error in compareParsers (line 17)
    r = RigidBodyManipulator(urdffile,struct('floating','rpy
```

https://drake-cdash.csail.mit.edu/testDetails.php?test=656025&build=19702

liangfok, 2016-05-14:
The failing unit test is declared in `[drake-distro]/drake/systems/plants/test/CMakeLists.txt`, and is implemented in `[drake distro]/drake/systems/plants/test/compareParsers.m`.

The failing unit test can be executed using the following commands:

```
$ cd [drake build artifacts directory]
$ ctest -VV -R systems/plants/test/compareParsers
```

I am currently unable to run the failing unit test because I'm encountering the following error, which is a known issue (https://github.com/RobotLocomotion/drake/issues/2194):

```
151: Error using allURDFs (line 20)
151: still need to implement a version that works without find
151:
151: Error in compareParsers (line 4)
151:   urdfs = allURDFs();
```

I will thus first address https://github.com/RobotLocomotion/drake/issues/2194 before further investigating this issue.

liangfok, 2016-05-17:
Now that https://github.com/RobotLocomotion/drake/pull/2356 was merged, I am able to re-test this again. Unfortunately, I wasn't able to replicate the error on my machine:

```
$ ctest -VV -R systems/plants/test/compareParsers
...
1/1 Test #151: systems/plants/test/compareParsers ...   Passed  100.01 sec

The following tests passed:
	systems/plants/test/compareParsers

100% tests passed, 0 tests failed out of 1

Total Test time (real) = 100.04 sec
```

liangfok, 2016-05-17:
I believe the error is stemming from:

> $ cd [drake distro]/drake/
> $ grep -r "Unrecognized file extension" .
./systems/plants/@RigidBodyManipulator/RigidBodyManipulator.m:          error('Drake:RigidBodyManipulator:UnsupportedFileExtension','Unrecognized file extension: %s for file %s',ext,name);

liangfok, 2016-05-17:
As a point for comparision, [here](https://drake-cdash.csail.mit.edu/testDetails.php?test=656025&build=19702) is the failure log and [here](https://drake-cdash.csail.mit.edu/testDetails.php?test=659103&build=21165) is a successful log.

liangfok, 2016-06-16:
I'm going to close this issue until we are able to link back to Jenkin's full console log from cdash and this problem occurs again. This is because I am unable to replicate this problem locally and the information in the CDash log doesn't provide sufficient insight into the root cause of the problem.

liangfok, 2017-01-11:
Again: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-continuous-matlab-open-source-debug/482/

liangfok, 2017-04-12:
The `compareParsers` unit test was removed in #4867.

closed 2017-04-12 by liangfok