RobotLocomotion/drake #2348: drake/common declarations should be in `drake::` namespace

sherm1, 2016-05-13:
I think that code that goes into the `drake/common` directory should reside in the `drake::` namespace. That seems sufficient to indicate that it is common across Drake. Putting it in `drake::common::` (which I did by rote for [NiceTypeName](https://github.com/RobotLocomotion/drake/blob/master/drake/common/nice_type_name.h#L11)) just mean a lifetime of unnecessary extra typing.

Does anyone think `drake::common::` adds anything useful over `drake::` here?

david-german-tri, 2016-05-13:
I agree.

liangfok, 2016-05-14:
I can think of equally strong arguments in either direction and will thus abstain.

jwnimmer-tri, 2016-05-16:
I can go along with this under two conditions, (1) no other directory populates `drake::` directly, and (2) given an identifier (`NiceTypeName` in this case) it is reasonably clear which `#include` statement it came from (true in this case).

PR #2358 (2016-05-16): Move NiceTypeName to drake:: namespace rather than drake::common
Resolves #2348 since this is currently the only thing in `drake/common`. 
Also fixed internal naming in test to match revised test name.

No feature changes; platform review please @jwnimmer-tri.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2358)
closed 2016-05-16 by jwnimmer-tri