RobotLocomotion/drake #2351: Enable NLOPT on Windows (and in Windows CI)

david-german-tri, 2016-05-14:
There is a third-party CMake file that might be useful.

http://ab-initio.mit.edu/wiki/index.php/NLopt_on_Windows

jamiesnape, 2016-05-17:
Is this being handled by the optimization team (per the label)? What is the priority? It would probably be best to split the CI part into a different ticket and assign it to me.

sammy-tri, 2016-05-18:
It's not currently bring handled by the optimization team.  We've actually found some fairly major issues with NLopt, so it might be best to wait on this one until we've found the right strategy going forward.

jamiesnape, 2016-05-18:
That is fine. Just checking that you were not waiting on us for any reason.

ggould-tri, 2016-05-25:
Dropping this to backlog due to more general nlopt issues.  If some future nlopt version is more promising, or if we can bound the bad behaviour classes of optimization problem, we can revisit it.

sammy-tri, 2016-09-12:
Closing since windows support is no longer an issue.

closed 2016-09-12 by sammy-tri