RobotLocomotion/drake #2352: Why is SNOPT disabled in Windows CI?

david-german-tri, 2016-05-14:
https://github.com/RobotLocomotion/drake-ci/blob/master/ctest_driver_script.cmake#L522

There's no corresponding condition in the Drake superbuild proper:

https://github.com/RobotLocomotion/drake/blob/master/CMakeLists.txt#L118

jamiesnape, 2016-05-16:
Because we cannot enable private repos on Windows due to a bug in Jenkins.

david-german-tri, 2016-05-18:
Proposed workaround from f2f discussion: bake SNOPT into the Windows AMI for now.

david-german-tri, 2016-06-08:
Update: @jamiesnape is still looking into working around the Jenkins bug to use SNOPT as a git external, but will time-box that at one day and dump it in the image instead.

jamiesnape, 2016-06-13:
Bug has been worked around, but now blocked by some sort of resolution of #2554.

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri