RobotLocomotion/drake #2357: WITH_LCM = OFF leads to build error

avalenzu, 2016-05-15:
When LCM support is disabled, `drakeRBSystem` is not built (see [this line](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/CMakeLists.txt#L144)), `drakeRBSystem_export.h` is not generated, and compilation of `RigidBodyTree.cpp` fails with

```
In file included from <path-to-drake-distro>/drake/systems/plants/RigidBodyTree.cpp:3:
In file included from <path-to-drake-distro>/drake/../drake/systems/plants/RigidBodyTree.h:17:
<path-to-drake-distro>/drake/../drake/systems/plants/RigidBodyFrame.h:6:10: fatal error: 'drake/drakeRBSystem_export.h' file not found
#include "drake/drakeRBSystem_export.h"
```

This error occurs because `drakeRBSystem` is not currently listed as a dependency for `drakeRBM`. I think we should keep it that way and eliminate the need to include `drakeRBSystem_export.h` in `RigidBodyFrame.h`.

liangfok, 2016-05-16:
I also just accidentally encountered this issue on my machine, except it's not because I set `WITH_LCM = OFF`. Instead, for some reason, LCM was not found on my machine when I tried to compile Drake this morning:

```
-- checking for module 'lcm'
--   package 'lcm' not found
-- disabling LCM.  lcm types will not be built.
```
Later, as this issue states, there's a compiler error due to a missing `drakeRBSystem_export.h`.
```
[ 26%] Building CXX object systems/plants/CMakeFiles/drakeRBM.dir/RigidBodyTree.cpp.o
In file included from /home/liang/dev/drake/drake/../drake/systems/plants/RigidBodyTree.h:15:0,
                 from /home/liang/dev/drake/drake/systems/plants/RigidBodyTree.cpp:1:
/home/liang/dev/drake/drake/../drake/systems/plants/RigidBodyFrame.h:5:40: fatal error: drake/drakeRBSystem_export.h: No such file or directory
 #include "drake/drakeRBSystem_export.h"
```

closed 2016-05-18 by david-german-tri

liangfok, 2016-05-25:
FTR, I now believe the lcm-not-found problem that I mentioned above is a manifestation of https://github.com/RobotLocomotion/drake/issues/2341.

PR #2391 (2016-05-18): Include the correct export file in RigidBodyFrame.h.
Closes #2357 (but does not resolve the various issues in which LCM is unexpectedly absent)

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2391) &emsp; Multiple assignees:&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/avalenzu">avalenzu</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/jwnimmer-tri">jwnimmer-tri</a>