RobotLocomotion/drake #2361: Expose Bullet (or any other collision library) specific API to user

amcastro-tri, 2016-05-16:
## The problem
[In my fork](https://github.com/amcastro-tri/drake/blob/APIAccesstoBulletModel/drake/systems/plants/test/lidarTest.cpp#L52) I show how it would be useful to expose Bullet specific (Bullet in this example but it could be FCL or any other) API's to the user. In that example the user has the option to specify a Bullet specific algorithm for ray casting. 

This would require the user to include `bullet_model.h` in their application as shown at [the top of the file in the same example in my fork](https://github.com/amcastro-tri/drake/blob/APIAccesstoBulletModel/drake/systems/plants/test/lidarTest.cpp#L16).

This approach fails however because [`bullet_model.h`](https://github.com/amcastro-tri/drake/blob/APIAccesstoBulletModel/drake/systems/plants/collision/bullet_model.h) includes all the Bullet headers polluting the user's application.  

## A solution 
Ideally `bullet_model.h` would not include any Bullet headers and it would only include the definition for the interface of a collision model that uses Bullet internally for collision handling. 
To accomplish this a solution would be to use a PIMPL implementation of `DrakeCollision::BulletModel`. In this way it is possible to split interface (no Bullet dependencies at all) from implementation (compiled with Bullet dependencies).

sherm1, 2016-05-16:
My first thought is that this is not a good idea because it exposes the fact that we are using Bullet, which we would like to replace with our own library later. If there is some option that a user needs to control, why not abstract that into Model and leave for the underlying code how to implement that option?

amcastro-tri, 2016-05-16:
Because these options are specific to the particular implementation. For instance FCL won't have the option to choose the `GJK` cast algorithm (and maybe others). That is why the user guards the code with `#ifdef BULLET_COLLISION` and then `dynamic_cast`'s to `BulletModel`.

sherm1, 2016-05-16:
Why should a user even have that option? Shouldn't they just ask us to raycast and let us figure out the best way to accomplish it with the collision engine we're using? For example, in this case we certainly don't want the user picking Bullet's broken algorithm.

closed 2017-02-01 by amcastro-tri