RobotLocomotion/drake #2366: Matlab License Error

liangfok, 2016-05-16:
I'm encountering the following problem:

```
Unrecognized MATLAB option "wait".
License checkout failed.
License Manager Error -97
License Manager cannot start. 
Check that the specified ports in the license file are not already in use. 
Restarting your machine may clear the ports.

Troubleshoot this issue by visiting: 
http://www.mathworks.com/support/lme/R2016a/97

Diagnostic Information:
Feature: MATLAB 
License path: /home/liang/.matlab/R2016a_licenses:/usr/local/MATLAB/R2016a/licenses/license.dat:/usr/local/MATLAB/
R2016a/licenses/network.lic 
Licensing error: -97,121.
```
@amcastro-tri also occasionally encounters this problem, which is sporadic and unpredictable.

david-german-tri, 2016-05-16:
This is a TRI-internal IT issue, not a Drake issue.

closed 2016-05-16 by david-german-tri

liangfok, 2016-05-16:
Should Drake's build system automatically switch to non-Matlab Drake when it encounters a problem with the local Matlab installation?

Perhaps print a warning prior to the switch?

I ask because I was just trying to build the documentation when this error popped up.

jwnimmer-tri, 2016-05-16:
If you've asked to have a particular working copy be a matlab build, and then matlab stops working, I think its entirely reasonable for the build / build system to complain and fail.  Build systems that say "oh well" and ignore problems are _not_ good build systems.  The textual output doesn't matter (so the warning text doesn't help); the only relevant output of a build should be its exitcode.

liangfok, 2016-05-16:
I don't mean to cause a ruckus but I don't think I "asked to have a particular working copy be a matlab build." Instead, the build system decided for itself to be a matlab build presumably because it found MATLAB in the path. Perhaps this brings about a bigger question of how much control we should have over the build settings.

jwnimmer-tri, 2016-05-16:
Yup.  And if you don't want to have matlab on by default, you can run `make options` again either with matlab unfindable on your path, or with `DISABLE_MATLAB=True` set.

liangfok, 2016-05-16:
Yeah, I'll use that. I was just thinking that if the build system can decide by itself to use matlab, it should also be able to decide by itself not to use matlab if it encounters a problem.