RobotLocomotion/drake #2369: ROS Indigo Needs to be Installed on CI Build Sever gcc + Linux

liangfok, 2016-05-16:
The official instructions for installing ROS Indigo are available [here](http://wiki.ros.org/indigo/Installation/Ubuntu). I've reproduced the commands that need to be executed below.

```
$ sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
$ sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net --recv-key 0xB01FA116
$ sudo apt-get update
$ sudo apt-get install ros-indigo-desktop
$ sudo rosdep init
$ rosdep update
$ echo "source /opt/ros/indigo/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
```
This issue must be resolved before we can resolve https://github.com/RobotLocomotion/drake/issues/2116.

jamiesnape, 2016-05-19:
Note that on the build server we would omit
```
$ echo "source /opt/ros/indigo/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
```
Instead, `source /opt/ros/indigo/setup.bash` would be added to the driver script in drake-ci. Shell startup scripts are not called by Jenkins.

david-german-tri, 2016-06-01:
As discussed f2f: `linux_ros` should be another column in the experimental build.  
@liangfok, after this is running, could you please document the install steps and update the supported platforms matrix?

liangfok, 2016-06-01:
OK will do.

david-german-tri, 2016-06-01:
Also as discussed f2f with Kitware: Try postsubmit first, before graduating this to presubmit.  There is already code that tries to use ROS, and we don't know whether it will explode.

jamiesnape, 2016-06-01:
@liangfok Is `ros-indigo-ros-base` sufficient or do you need a full desktop install?

liangfok, 2016-06-02:
@jamiesnape: Good question. There are a couple of additional packages required. Here is the  installation command:

```
$ sudo apt-get install ros-indigo-ros-base ros-indigo-tf  ros-indigo-ackermann-msgs 
```
Things may change as I continue to refine on the Drake / ROS integration but at least the above will allow us to compile and run what's currently implemented.

jamiesnape, 2016-06-03:
Thanks. I will roll out a new image with those packages installed in the next day or so. Just create a ticket and ping me if you need more packages installed in future.

liangfok, 2016-06-03:
Thanks! Will do. I will be submitting PRs that contain ROS-based unit tests once the new image with ROS is available.

liangfok, 2016-06-08:
@jamiesnape: For simple unit tests like the current `ros_test.cc` within Drake, we _could_ make it a unit test using CMake's `add_test(...)` and run it as long as `roscore` can started as a separate process prior to the execution of the unit test and shut down after the execution of the unit test. I'm not sure if this is worthwhile doing as a temporary workaround until we're able to modify Drake's build system to fake a Catkin workspace. The ability to fake a Catkin workspace is preferable because it will allow us to run more sophisticated unit tests that involve multiple ROS nodes (each being a separate process).

jamiesnape, 2016-06-14:
ROS is now installed on the Linux image. Next, I need to configure a job that uses it.

liangfok, 2016-06-14:
That's great! Thanks!

I'd ideally like to "fake" a Catkin package and run rostests as described here: https://github.com/RobotLocomotion/drake/issues/2259#issuecomment-224089576

I'm planning on being in CAM3 soon. Maybe we can discuss f2f.

jamiesnape, 2016-06-14:
Sounds good.

jhoare, 2016-06-22:
https://github.com/RobotLocomotion/drake-ci/pull/6 adds the ``setup.bash`` environment variables to the CI.

jamiesnape, 2016-07-18:
ROS Indigo is installed.

closed 2016-07-18 by jamiesnape