RobotLocomotion/drake #2383: Dirtran bug using midpoint interpolation?

kuindersma, 2016-05-17:
I was just looking at the dirtran Matlab implementation and it seems that we should be constraining u(N)=0 when using midpoint interpolation since u(N) affects x(N) but is never executed.

RussTedrake, 2016-05-25:
maybe not =0 (in case 0 is not in the feasible u), but something similar.  @mposa, @hongkai-dai -- can you give @kuindersma a sanity check on this?

hongkai-dai, 2016-05-25:
Maybe we should use the u in the steady state, if we want the trajectory to stay in equilibrium in the end?

I do not fully understand this, although u(N) is not executed in the controller, the interpolated control action between u(N-1) and u(N) will still be executed, and u(N) does affect those interpolated control actions, so the value of u(N) should matter for the controller?

kuindersma, 2016-05-25:
So you're saying you always create a new utraj that does midpoint interpolation on the utraj that is output from DirtranTrajectoryOptimization before using it in a controller? If I understand that correctly, then maybe the solution is to override reconstructInputTrajectory in DirtranTrajectoryOptimization for use with MIDPOINT to keep the contract the same as the other interpolation schemes?

hongkai-dai, 2016-05-25:
You are correct, the reconstruction of the control trajectory is not right. Actually there is an issue #509 opened a while ago, about reconstructing the control trajectory.

hongkai-dai, 2016-05-25:
Sorry for being really sluggish to fix that issue #509. I will work on it tonight.

kuindersma, 2016-05-25:
No rush from me as long as its a known issue. Closing as duplicate.

closed 2016-05-25 by kuindersma