RobotLocomotion/drake #2384: Iris fails to configure when MOSEK is unavailable

jamiesnape, 2016-05-17:
```
Performing configure step for 'iris_project'
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /usr/bin/gcc-4.9
-- Check for working C compiler: /usr/bin/gcc-4.9 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++-4.9
-- Check for working CXX compiler: /usr/bin/g++-4.9 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found matlab
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- Found eigen3
-- Checking for module 'mosek'
--   No package 'mosek' found
-- Configuring incomplete, errors occurred!
```
https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=21741

jamiesnape, 2016-05-17:
Iris will be disabled from the superbuild at 3:30pm ET if issue is still outstanding.

david-german-tri, 2016-05-17:
Isn't it already disabled in the superbuild?

jamiesnape, 2016-05-17:
It is enabled on MATLAB builds other than Windows.

rdeits, 2016-05-17:
iris requires mosek, so we shouldn't be enabling it on systems that don't have mosek enabled

jamiesnape, 2016-05-17:
@rdeits You may wish to update the documentation of Iris. It does not say MOSEK is required.

jamiesnape, 2016-05-17:
I will disable on CI. Someone will need to follow up in the main repo so that Iris cannot be enabled without MOSEK.

PR #2463 (2016-05-31): Infrastructure for declaring MATLAB dependencies for ctest
Addresses in part #1997. Someone will need to go through all the MATLAB files and correlate the dependencies that are required for each `ctest` test. A limited number have been declared to test this PR and as an example. Also fixes #2384.

FYI @RussTedrake.

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2463)
closed 2016-05-31 by jamiesnape