RobotLocomotion/drake #2385: Malformed TOC Entry on mac build

jonathan-owens, 2016-05-17:
```
ld: malformed archive TOC entry for _bot_core_vector_3d_t_unsubscribe, offset 808216 is beyond end of file 516096
 file '/Users/yosemite/workspace/mac-clang-continuous-debug/build/lib/liblcmtypes_bot2-core.a' for architecture x86_64
ld: malformed archive TOC entry for _bot_core_vector_3d_t_unsubscribe, offset 808216 is beyond end of file 516096
 file '/Users/yosemite/workspace/mac-clang-continuous-debug/build/lib/liblcmtypes_bot2-core.a' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[8]: *** [bin/bot-param-tool] Error 1
make[8]: *** [bin/bot-param-server] Error 1
make[7]: *** [src/param_server/CMakeFiles/bot-param-server.dir/all] Error 2
make[7]: *** Waiting for unfinished jobs....
make[7]: *** [src/param_server/CMakeFiles/bot-param-tool.dir/all] Error 2
make[6]: *** [all] Error 2
make[5]: *** [bot2-param-prefix/src/bot2-param-stamp/bot2-param-build] Error 2
make[4]: *** [CMakeFiles/bot2-param.dir/all] Error 2
```

https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=21971

jonathan-owens, 2016-05-18:
Still happening: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=22412

jonathan-owens, 2016-06-01:
It seems that this is caused by a stale library existing somewhere. See [here](http://trac.wildfiregames.com/ticket/1147). I cannot reproduce this locally. My best guess would be that some changes introduced that triggered the build affected the libraries and those ``.a`` files were not properly updated.

I am trying to take a look at the code that triggered those builds, but can't seem to find a way to link the dashboard build back to the changes. @jamiesnape is there a way to do this? Say, for build #21971.

jamiesnape, 2016-06-01:
The OS X builds always happen on a completely fresh VM, so not sure why there would be a stale library. To get changes go to

https://drake-jenkins.csail.mit.edu/job/mac-clang-continuous-debug/184/changes

The changed code does not run on that build, though.

jonathan-owens, 2016-06-01:
That's interesting. I thought it started fresh, but all I can find anywhere online is to delete the offending archive file and try again. I'm not sure what else to do at this point. Perhaps we can close this for now and I will keep an eye on it, especially as I am build cop for the next couple of days, anyway. What do you think?

jamiesnape, 2016-06-01:
Sounds fine.

jamiesnape, 2016-06-02:
Error is back again: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=30608

jonathan-owens, 2016-06-22:
This continues: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=38598

liangfok, 2016-06-24:
Here too: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=40617

david-german-tri, 2016-10-12:
Maybe it's gone now?  Closing.

closed 2016-10-12 by david-german-tri