RobotLocomotion/drake #2390: Use `-nojvm` option when starting MATLAB on Linux and OS X?

jamiesnape, 2016-05-17:
> Mac and Linux® platforms
> `-nojvm`
> Start MATLAB without loading the JVM™ software. This minimizes memory usage and improves initial startup speed, but restricts functionality. With nojvm, you cannot use the desktop, figures, or any tools that require Java® software.

-- http://www.mathworks.com/help/matlab/matlab_env/commonly-used-startup-options.html

We start MATLAB >400 times per build. Do we require any functionality that is restricted by disabling the JVM?

david-german-tri, 2016-05-18:
Examples often invoke figures, so it depends how hard the failure is with `-nojvm`.  As we just discussed f2f, I'll give it a try (once I dig out from my notes how to pass a flag to MATLAB).

david-german-tri, 2016-05-18:
I'm going to stream notes to this issue for a little bit; no need for anyone to reply.

I got errors around here on my first attempt to use `-nojvm`.  Don't yet know if they're fatal or not, but the comment implies we need the JVM for at least some tests.

https://github.com/RobotLocomotion/drake/blob/59a62d2dbe484b713c6e25561bdc3d647a4d17a7/drake/addpath_drake.m#L94

david-german-tri, 2016-05-18:
@RussTedrake says he put a fair amount of work into stripping out Java dependencies in the past, in the hopes that `-nojvm` would improve performance on Atlas.  However, he was never able to get completely Java-free.  So, I'm going to conclude the answer to the original question is "yes", and close this issue.  Too bad.  :-/

closed 2016-05-18 by david-german-tri