RobotLocomotion/drake #2393: Drake Doxygen Website Does Not Include Examples

liangfok, 2016-05-18:
It would be nice to generate the Doxygen webpages for the classes within `drake/examples`. At a minimum, this would enable us to verify that our Doxygen comments are rendered correctly.

liangfok, 2016-06-07:
I traced the problem to: https://github.com/RobotLocomotion/drake/blob/master/drake/doc/Doxyfile_CXX.in#L128

sherm1, 2016-06-07:
I think this could be confusing, Liang. The examples are likely to have many repeated symbol names. Nice doxygen could be made for them but not the way we usually do it -- I think they would need to have their own topics and be documented consciously as examples.

liangfok, 2016-06-07:
Yeah I was just thinking the same. Maybe each example can have its own Doxygen website that is built using:

```
$ make documentation-[example name]
```

sherm1, 2016-06-07:
I think the example documentation should just get built along with Sphinx and the API documentation -- it's too much work to ask for manual builds of each example.

liangfok, 2016-06-07:
OK about generating both Drake's library API and its examples using the same build target and command. How about we enable example documentation when [INTERNAL_DOCS = YES](https://github.com/RobotLocomotion/drake/blob/master/drake/doc/Doxyfile_CXX.in#L75) (it will be by default set to `NO`)?

jwnimmer-tri, 2016-12-22:
Possibly fixed?

RussTedrake, 2016-12-23:
yes.  examples are built in the main documentation.

closed 2016-12-23 by RussTedrake