RobotLocomotion/drake #2397: CI clang-mac Problem - "Agent went offline during the build"

liangfok, 2016-05-19:
Here's an example: 

https://drake-jenkins.csail.mit.edu/job/experimental/1562/compiler=clang,label=mac/console

Any idea what's the cause?

david-german-tri, 2016-05-19:
@jamiesnape FYI we're still seeing these occasionally post-XServe.

jamiesnape, 2016-05-19:
Network issues, especially as we are over-capacity due to maintenance. Whatever type of Mac we use, it will always be located 900 miles away through a VPN connection.

jamiesnape, 2016-05-19:
By all means file the issue, but I cannot fix the realities of cloud computing.

david-german-tri, 2016-05-19:
That makes sense.  Thanks.

liangfok, 2016-05-19:
@jamiesnape: Thanks for the explanation.
@david-german-tri Thanks for closing the issue now that we have an explanation.

liangfok, 2017-01-06:
I just encountered this problem; first time not on an OSX CI instance:

```
14:12:47 207/208 Test #108: Agent went offline during the build
14:28:53 ERROR: Connection was broken: java.io.IOException: Unexpected termination of the channel
14:28:53 	at hudson.remoting.SynchronousCommandTransport$ReaderThread.run(SynchronousCommandTransport.java:73)
14:28:53 Caused by: java.io.EOFException
14:28:53 	at java.io.ObjectInputStream$PeekInputStream.readFully(ObjectInputStream.java:2351)
14:28:53 	at java.io.ObjectInputStream$BlockDataInputStream.readShort(ObjectInputStream.java:2820)
14:28:53 	at java.io.ObjectInputStream.readStreamHeader(ObjectInputStream.java:804)
14:28:53 	at java.io.ObjectInputStream.<init>(ObjectInputStream.java:302)
14:28:53 	at hudson.remoting.ObjectInputStreamEx.<init>(ObjectInputStreamEx.java:48)
14:28:53 	at hudson.remoting.AbstractSynchronousByteArrayCommandTransport.read(AbstractSynchronousByteArrayCommandTransport.java:34)
14:28:53 	at hudson.remoting.SynchronousCommandTransport$ReaderThread.run(SynchronousCommandTransport.java:59)
```
https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental-open-source-debug/2161/

jamiesnape, 2017-01-06:
Even Amazon cannot guarantee 100% uptime on instances.

liangfok, 2017-01-06:
Yeah, I suppose expecting 100% server reliability is too much. I will close this issue for now.

closed 2017-01-06 by liangfok