RobotLocomotion/drake #2398: Windows builds failing functional_form_test

jonathan-owens, 2016-05-19:
See, e.g. https://drake-cdash.csail.mit.edu/testDetails.php?test=701236&build=22748.

@bradking: The below changes seem to have broken the Windows builds.

https://github.com/RobotLocomotion/drake/commit/ab8bf49c61cec1192b6b179f3de4a136cc7ccd1f

bradking, 2016-05-19:
See PR #2400.

bradking, 2016-05-19:
Strange that it worked locally and in all the pre-integration tests.  I suspect the difference is whether `PATH` in the environment contains `$top/drake/lib/$config` in addition to `$top/install/lib`.  With both entries then the install rule is not needed to run the test.

billhoffman, 2016-05-19:
I don't think it is a PATH issue unless this test requires a different PATH than any of the other tests.

billhoffman, 2016-05-19:
It is just failing on the nightly build: https://drake-cdash.csail.mit.edu/testSummary.php?project=1&name=functional_form_test&date=2016-05-19

bradking, 2016-05-19:
> I don't think it is a PATH issue

My PR fixes it, but I'm wondering how this was passing anywhere before the fix.

billhoffman, 2016-05-19:
Seems like the ctest script has too much in the PATH... It has this:     "${DASHBOARD_WORKSPACE}/drake/pod-build/lib/Release"  That would explain why it failed on MS IDE Debug, and all Ninja builds, but passed on all release builds.

bradking, 2016-05-19:
> Seems like the ctest script has too much in the PATH

Yes, that should be removed so all the builds work consistently.

billhoffman, 2016-05-19:
Yup, we are removing it.

bradking, 2016-05-19:
With PR 2400 merged this issue is resolved.

closed 2016-05-19 by bradking