RobotLocomotion/drake #2410: Strange Matlab Failures

liangfok, 2016-05-19:
See: https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=23199

Which is part of: https://drake-jenkins.csail.mit.edu/job/experimental/1585/compiler=gcc,label=linux_matlab/

david-german-tri, 2016-05-25:
Downgrading to backlog, consistent with the 2016-05-25 tracker scrub decision that all flaky MATLAB test failures are backlog-priority.

jwnimmer-tri, 2016-07-19:
Two months old with no further recurrences posted.  @david-german-tri or @liangfok can we just close this?

liangfok, 2016-07-19:
Yes, we can close it. I updated the name so we can more easily dig it up again in case the error comes up again.

jonathan-owens, 2016-11-28:
It seems that this is still happening, as well as with a (I believe) similar error, cell2mat:

https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=218965

liangfok, 2016-11-28:
Normally, I would think that these errors are due to MATLAB toolbox license issues except `num2cell` and `cell2mat` appear to be functions that are part of MATLAB itself (and not part of toolbox)!

Lucy-tri, 2016-11-30:
Undefined function 'num2cell' for input arguments of type 'double' happened here: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-continuous-matlab-open-source-debug/236/
I'm not sure which link is best:
https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-linux-gcc-continuous-matlab-open-source-debug-236&field2=buildstarttime&compare2=84&value2=now[](url)

hongkai-dai, 2016-11-30:
> 20Production/job/linux-gcc-continuous-matlab-open-source-debug/236/
> showfilters=1&filtercount=2&showfilters=1&filtercombine=
> and&field1=label&compare1=61&value1=jenkins-linux-gcc-
> continuous-matlab-open-source-debug-236&field2=buildstarttime&compare2=84&
> value2=now[](url)

The error happens in this line

Error in RigidBodyManipulator/adjustCollisionGeometry (line 2374)

[num2cell(find(body_changed))';getLinkName(model,body_changed)];

The reason is that num2cell expect a vector (not a scalar). So previously
after we compile the collision geometry, multiple bodies are changed. Now
for some reason only one body is changed.

Lucy-tri, 2016-11-30:
@hongkai-dai  I looked for possible source code changes, but I didn't see code changes in that build, and no recent code changes in Matlab (the code you mention is in RigidBodyManipulator.m). And the next build passed the test. So I can't understand how the Matlab code could be doing something different than it used to.

Lucy-tri, 2016-12-01:
Undefined function 'num2cell' for input arguments of type 'double'.
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-gcc-ninja-nightly-matlab-debug/121/

Lucy-tri, 2016-12-01:
Undefined function 'sub2ind' for input arguments of type 'double'.
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-gcc-ninja-nightly-matlab-release/194/

Lucy-tri, 2016-12-02:
It's true that a problem with Matlab (licensing?) can result in undefined functions, but then there will be dozens of such failures, as in this build which as 62 failures:  https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-ninja-continuous-matlab-open-source-debug/238/
Is it possible to have a Matlab licensing failure just for a second or so?

jwnimmer-tri, 2016-12-02:
Yes, sure.

jonathan-owens, 2017-03-03:
Looks like this (and a slight variation?) are still happening:

https://drake-cdash.csail.mit.edu/testDetails.php?test=9097912&build=434644
and
https://drake-cdash.csail.mit.edu/testDetails.php?test=9070385&build=433326

liangfok, 2017-03-31:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-continuous-matlab-open-source-debug/939/

david-german-tri, 2017-04-13:
MATLAB license failures are well understood and not fixable.  We don't need a tracking issue for this.  Closing.

closed 2017-04-13 by david-german-tri