RobotLocomotion/drake #2413: Improvements to jenkins build of drake documentation

jwnimmer-tri, 2016-05-20:
(Broken out of #2403 discussion.)

For documentation-only PRs, it would be nice to have the fastest possible pass / fail result from jenkins (whether the doc changes broke the build), but I believe currently the documentation build is configured as part of the matlab build, and waiting for matlab tests (or any tests) to run first seems like it will would slow things down somewhat.

I suppose running the doc build before ctest, and then stopping the run early if docs failed, just might be good enough?

It would be clearest to me if jenkins had a specific "docs" build that only build the docs, but perhaps the cost of spinning up an instance just to do that means it wouldn't be faster, just clearer.

So assuming that there isn't going to be a "doc" build variant in jenkins (i.e., that docs are rolled into some other build), it would also be nice if somewhere in our drake docs (developers.rst?) we advised developers where in jekins their docs are going to be built, so they can click through to that item directly, and so that platform reviewers (mergers) know which jenkins items are relevant to doc-only PRs, so we can merge before all variants finish building.

jamiesnape, 2016-05-20:
We really could do with cleaning up the Sphinx and Doxygen warnings so we can switch on the equivalent of `-Werror` for those. See #2408 and #2409.

jwnimmer-tri, 2016-05-20:
Agree for sphinx.  Doxygen I've been burned before, so would need to ponder a bit more.

sherm1, 2016-05-20:
> Agree for sphinx. Doxygen I've been burned before, so would need to ponder a bit more.

I agree for both. I've had good luck with large projects with a warning-free Doxygen requirement. In my experience the warnings are almost always actual problems. However, currently we are very close for Sphinx and very far away for Doxygen. (I think this is because we haven't been requiring developers to test their Doxygen along with their code. Ahem.) We could fix Sphinx in one PR and stretch out the Doxygen repairs for a while.

jwnimmer-tri, 2017-01-17:
Possible that #3129 offers an opportunity to offer a warning- or error-sensitive label to run the doc builders, and that we could have a doc-only build on PRs that posts results for browsing.

EricCousineau-TRI, 2018-11-02:
Given that we have our specific `-documentation` jobs on Jenkins, I've scoped this issue to build + test Doxygen under `bazel test //...`. We can reword if necessary.

jamiesnape, 2018-11-02:
Possible still an open issue about warning-free Doxygen. Malformed Doxygen is of slightly more consequence now that we try to parse it for the Python docs too.

jwnimmer-tri, 2018-11-02:
I would actually proposed that we can close this.  I don't think there have been many (or any?) PRs in recent memory that have passed the existing pre-merge CI suite, but that then failed on the documentation nightly.  If we see nightlies failing (for this, or any other reason), we can work to mitigate it in continuous or experimental, on an as-needed basis.

jamiesnape, 2018-11-02:
We don't -Werror on the Doxygen, so there is not really anything for it to choke on.

jwnimmer-tri, 2018-11-02:
Exactly.  So there isn't a pain point related to quick-turnaround docs-only CI builds, as was hypothesized in my opening post atop this issue.  If you guys want an issue about turning on Werror for Doxygen, I think that's a new ticket (and, as above, I still remain skeptical that it's worth our investment -- really ever, and in any case not right now).

closed 2018-11-02 by jwnimmer-tri