RobotLocomotion/drake #2416: MATLAB tests fail when drake\pod-buildlib/Release is not in PATH

jamiesnape, 2016-05-20:
https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=23667

RussTedrake, 2016-05-20:
Could to catch and flag these.  I know that we were not sufficiently careful about this in the past.

RussTedrake, 2016-05-25:
@jamiesnape -- totally agree this is a bug.  do you have someone that can investigate and resolve?

RussTedrake, 2016-05-25:
when resolved, we should also be able to remove this line:
https://github.com/RobotLocomotion/drake/blob/master/drake/addpath_drake.m#L90
(in fact, I'm not sure how the matlab tests are failing, since this line should cover that case)

closed 2016-05-26 by jamiesnape

david-german-tri, 2016-06-01:
Closing the loop on an offline conversation related to #2448: I don't think there's an analogy between this discussion and C++ linking.  It's normal and necessary for some build targets to link against libraries that aren't installed.

jamiesnape, 2016-06-01:
Of course, but in the world of PODs apparently everything should be installed and have a pkg-config file.

RussTedrake, 2016-06-01:
That’s not necessarily true of PODs.  Only libraries that are shared across pods should be installed.

david-german-tri, 2016-06-01:
PODs: we will not miss them.

jamiesnape, 2016-06-01:
That was not the case here either before or after #2455. Either way, it should be tidier if/when we follow the normal CMake patterns.

PR #2455 (2016-05-26): Add missing pkg-config files
Fixes #2416.

* https://drake-jenkins.csail.mit.edu/job/windows-msvc-ninja-32-experimental-matlab/4/
* https://drake-jenkins.csail.mit.edu/job/windows-msvc-ninja-64-experimental-matlab/7/

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2455)