RobotLocomotion/drake #2422: C++ Methods Are Not Included in Doxygen Website

liangfok, 2016-05-21:
For example, I'm not seeing `parseWorldJoint()`, which is defined in `drake/systems/plants/RigidBodyTreeURDF.cpp`. 

By not showing up, it's hard to test / debug the Doxygen comment.

sherm1, 2016-05-21:
Doxygen comments should be collected only from header files, not `.cpp` or `.cc` files. Any doxygen in the code should be moved from the definition to the header declaration if it is intended to be visible.

liangfok, 2016-05-21:
I would argue that Drake developers would be interested in seeing _every_ method documented in Doxygen. We can later remove those that are not relevant to Drake users when we generate a Doxygen website for Drake users.

If no one objects, I will start adding method declarations to header files for methods that currently don't show up in the Doxygen-generated website.

sherm1, 2016-05-21:
I'm not sure exactly what you mean, Liang. Doxygen documentation is for API users and should not confuse them with anything that is not part of the intended API. There are other modes you can run Doxygen in to produce output that is useful for developers -- you can even make it document private methods. But that is inappropriate for end users and should not be out normal use of Doxygen. You would need to make a special developer's Doxyfile to produce an alternate set of documents.

liangfok, 2016-05-21:
We have not formally defined who are audience is. Based on informal conversations, I believe we are currently targeting domain experts who are also c++ experts. These people, in my mind, would all fall under the category of "Drake developers". It actually excludes "Drake users". Thus, I'm thinking our Doxygen output should target people who are working deeply within Drake's implementation.

Once Drake has a stable API (maybe by Q4?) we can start tweaking the Doxygen website by removing methods that we believe are of no interest to a typical "Drake user."

sherm1, 2016-05-21:
I disagree. Doxygen should be for users. It takes a long time to develop good API documentation and we should be doing it and debugging it now. Most Drake users, including Russ's Underactuated Robotics students, access Drake through Matlab and use the Matlab-formatted doxygen that comes from the same source code as our C++ doxygen. Developers are very resourceful and can read the code or generate their own customized documentation.

liangfok, 2016-05-21:
Oh, I didn't realize that changing the C++ Doxygen results in chaining the Matlab Doxygen. Is that true? If so, I would consider that a defect.

liangfok, 2016-05-21:
Here is another consideration:

C++ Drake is still in a nascent form that will continue to undergo tremendous change in the near future. We are not entirely sure how people will be using C++ Drake in the future. Thus, wouldn't it be presumptuous of us to assume that certain methods within Drake will not be of use to any future C++ Drake user? Wouldn't it be better to expose every single method at least initiallyou and then, after several million users exist, we can ask "so which methods are not being used by anybody?". These methods can then be removed from the public API.

sherm1, 2016-05-22:
APIs have to be carefully designed. Throwing everything in and then removing what doesn't stick is not a good approach!

liangfok, 2016-05-22:
OK. I will close this issue for now.

closed 2016-05-22 by liangfok