RobotLocomotion/drake #2427: Documentation SSL Certificate Error

wxmerkt, 2016-05-23:
Given that the documentation is hosted on GitHub Pages and a custom subdomain is being used, SSL certificate errors are shown when accessing https://drake.mit.edu as the certificate is for *.github.io/github.com. 

GitHub Pages [does not support HTTPS](https://github.com/isaacs/github/issues/156) just yet, but there are workarounds using a [reverse proxy](https://pascal.io/github-pages-https/) or [CloudFlare](https://blog.keanulee.com/2014/10/11/setting-up-ssl-on-github-pages.html).

jamiesnape, 2016-05-23:
The URL is http://drake.mit.edu. Where are you seeing a reference to HTTPS?

david-german-tri, 2016-05-23:
+1 for @jamiesnape's question - if there are references to HTTPS in Drake docs, we should convert them to HTTP, but I don't think it's worth maintaining a proxy workaround.

wxmerkt, 2016-05-23:
When I've searched specific documentation for Drake on Google it defaulted to HTTPS

jamiesnape, 2016-05-23:
Google will correct itself within a few pages.

closed 2016-05-23 by wxmerkt