RobotLocomotion/drake #2428: Make disabling -Werror a CMake option

jamiesnape, 2016-05-23:
#2370 is breaking some of the more exotic CI builds, so we need an easy way to switch it off.

CC @david-german-tri @jwnimmer-tri.

jwnimmer-tri, 2016-05-23:
@jamiesnape Please link to one or two breaks, for our reference.

jamiesnape, 2016-05-23:
e.g.
```
option(MAKE_WARNINGS_INTO_ERRORS "Make warnings into errors" ON)
if(MAKE_WARNINGS_INTO_ERRORS)
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Werror")
endif()
```

jamiesnape, 2016-05-23:
The MSan builds, in particular. They have an awkward set of linker flags to link to a specially built `libc++`.

jamiesnape, 2016-05-23:
And it is sometimes useful when developing to have this option, anyway.

jwnimmer-tri, 2016-05-23:
[Jenkins msan](https://drake-jenkins.csail.mit.edu/view/Dynamic%20Analysis/job/linux-clang-nightly-memcheck-msan/) link for the record.

Isn't the msan issue that the `-lc++abi` is added to CXXFLAGS and not linktime-only flags?  What is the nominal CMake spelling for CXXLDFLAGS or whatever?

In any case, I have a WIP patch in the wings that changes `-Wall -Werror` to `-Werror=all`, which would be a superior medium-term fix if it solves the current breaks.  (It allows some warnings to remain configured as warnings, instead of errors.)  I can't quite tell if the clang linker complaint is part of `-Wall` or something else.

jamiesnape, 2016-05-23:
It is a pods workaround for the superbuild. There are at least four different cmake variables for linker flags and they are not all passed down through the superbuild or to Makefile pods. Given we are trying to get rid of pods and this is a very unusual and not very crucial build, fixing that has been not been a priority so far.

jwnimmer-tri, 2016-05-24:
Per a local spike test with `clang++-3.7`, changing from `-Wall -Werror` to `-Werror=all` is enough to demote the `clang-3.7: error: -lc++abi: 'linker' input unused` from an error back to a warning.  I am working up a PR to use that phrasing for both gcc and clang.

jwnimmer-tri, 2016-05-24:
@jamiesnape Per [nightly msan 107](https://drake-jenkins.csail.mit.edu/view/Dynamic%20Analysis/job/linux-clang-nightly-memcheck-msan/107/) that I launched by hand, it looks like we're back to "normal" now as far as the msan build warnings.

I'd like to thus propose closing this issue.  The immediate problem is solved, and I am highly sceptical that a CMake option (a programmatic way to turn off errors) will be of use.

closed 2016-05-26 by jamiesnape