RobotLocomotion/drake #2433: Error using copyfile in MATLAB

jonathan-owens, 2016-05-23:
```
[[NON-XML-CHAR-0x8]Warning: RigidBodyManipulator: velocity limits are not supported yet][NON-XML-CHAR-0x8] 
[[NON-XML-CHAR-0x8]> In WarningManager/warnOnce (line 21)
  In RigidBodyManipulator/addRobotFromURDF>parseJoint (line 357)
  In RigidBodyManipulator/addRobotFromURDF>parseRobot (line 105)
  In RigidBodyManipulator/addRobotFromURDF (line 55)
  In RigidBodyManipulator (line 75)
  In runPassive (line 12)][NON-XML-CHAR-0x8] 
Constructing a new airfoil
Error using copyfile
The process cannot access the file because it is being used by another process.

Error in RigidBodyWing/onCompile (line 137)
          copyfile(which(profile),fullfile(tempdir,[filename,fileext]));

Error in RigidBodyManipulator/compile (line 720)
        [new_element, model] = model.force{i}.onCompile(model);

Error in RigidBodyManipulator/addRobotFromURDF (line 60)
model = compile(model);  % ideally this would happen on entry into any function...

Error in RigidBodyManipulator (line 75)
          obj = addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);

Error in runPassive (line 12)
p = RigidBodyManipulator('Plane.URDF', options);

 

CMake Error at C:/projects/Agzs6sMa/drake/cmake/MatlabTestsRedirect.cmake:108 (message):
  [MATLAB] TEST FAILED Matlab returned 1

```
See, e.g., https://drake-cdash.csail.mit.edu/testDetails.php?test=727176&build=24472. Right now this is just on Windows.

RussTedrake, 2016-05-25:
that's super annoying. my best guess is that two parallel jobs (the two in https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Wingeron/CMakeLists.txt ) happened to hit this line at exactly the same time to cause the failure.  

@jonathan-owens -- is there a way to tell ctest "ok to run in parallel, but just don't run these two tests at the same time"?

RussTedrake, 2016-05-25:
actually, the better fix is to copy to a temp file name (not just a temp file directory).  took me a minute to even realize it wasn't that way.

commit 249381fa (2016-05-25): resolves #2433
commit d0347a5d (2016-05-25): Merge pull request #2449 from RussTedrake/copy_wing

resolves #2433
closed 2016-05-25 by david-german-tri