RobotLocomotion/drake #2435: Jenkins seems to be buliding origin/master instead of origin/pr/XXXX/merge

sammy-tri, 2016-05-23:
Jenkins builds on new PRs are building from origin/master instead of the merged PR.  Seems to have started sometime on the morning of the 23rd.

Build 1623 seems OK: [](https://drake-jenkins.csail.mit.edu/job/experimental/1623/ ) [](https://drake-jenkins.csail.mit.edu/job/experimental/1623/compiler=clang,label=linux/git/)

Build 1630 seems to be from master: [](https://drake-jenkins.csail.mit.edu/job/experimental/1630/) https://drake-jenkins.csail.mit.edu/job/experimental/compiler=clang,label=linux/1630/git/

jwnimmer-tri, 2016-05-23:
/cc buildcop @jonathan-owens

david-german-tri, 2016-05-23:
And /cc @jamiesnape who has requested to be pinged on all things CI.

david-german-tri, 2016-05-23:
`master` is the default value of the `sha1` parameter in the experimental build, but I think that's been true for a while...trying to confirm from the Jenkins config history, and to figure out how it's overridden on a per-build basis.

jamiesnape, 2016-05-23:
The GitHub pull request builder plugin sets $sha1.

jamiesnape, 2016-05-23:
It looked like someone manually triggered a job today too, for some reason, so $sha1 there would be whatever they entered.

jamiesnape, 2016-05-23:
https://github.com/janinko/ghprb#parameterized-builds

david-german-tri, 2016-05-23:
Another interesting difference between build 1623 and build 1630 is that 1623 has one instance of "Parameters" in its left menu, while 1630 has fourteen.

jamiesnape, 2016-05-23:
GitHub hooks have been flaky today too, since GitHub has had API problems.

david-german-tri, 2016-05-23:
GitHub is claiming their fires are out: https://status.github.com/.  I've kicked off a retest on #2431 to see if we're back to normal behavior.

jamiesnape, 2016-05-23:
Something is very wrong. We have 221 builds in the queue.

jamiesnape, 2016-05-23:
Up to 297. I am going to reboot Jenkins before it melts down.

sammy-tri, 2016-05-23:
And based on https://drake-jenkins.csail.mit.edu/job/experimental/compiler=cpplint,label=linux/1656/console, they're still not building the desired targets...

jamiesnape, 2016-05-23:
Some are and some aren't. Not sure why yet.

jamiesnape, 2016-05-23:
Something is definitely wrong with how parameters are passed down to the matrix.

jamiesnape, 2016-05-23:
The changelog is correct: https://drake-jenkins.csail.mit.edu/job/experimental/1713/

jamiesnape, 2016-05-23:
But the SHA is not passed down to the matrix builds.

jamiesnape, 2016-05-23:
https://issues.jenkins-ci.org/browse/JENKINS-34758

jamiesnape, 2016-05-23:
Compiling a version of the plugin with the fix now.

jamiesnape, 2016-05-23:
Looks like we are fixed now.

jamiesnape, 2016-05-24:
I am thinking it may be better to not have a default value for `$sha1` for these builds. Not many people manually run them and a failure would be preferable to building the wrong branch.

david-german-tri, 2016-05-24:
agreed

jamiesnape, 2016-05-24:
Done.

closed 2016-05-25 by david-german-tri