RobotLocomotion/drake #2436: Unlisted dependency on Python 3 numpy

geronm, 2016-05-23:
Unfortunately it wasn't until after my attempted fix that I thought to copy/paste the error message, but drake refused to build until I had installed the python3 version of numpy, throwing an error message along the lines of "Python error: the package 'Numpy' is not installed.".

`sudo pip install numpy` resulted in a message saying it was already installed, as did starting python and running `import numpy`. Only after I ran `sudo pip3 install numpy` did the make proceed.

I believe the error occurred in the vicinity of MATLAB Swig being installed; in any case, it should be straightforward to reproduce the message.

david-german-tri, 2016-11-04:
I don't know when it was fixed, but `numpy` now appears in the OS X and Trusty docs as well as the Xenial script.

closed 2016-11-04 by david-german-tri

jamiesnape, 2016-11-04:
https://github.com/RobotLocomotion/drake/commit/dafe7bc6cbc834de7e0a1f49d3cd3fc15b6c041 and https://github.com/RobotLocomotion/drake/commit/f42d6f2c9e6d77b15ae31ba2efc422d0102debac (March 23 / April 14).