RobotLocomotion/drake #2450: Add more realistic collision models to Prius.sdf / urdf

liangfok, 2016-05-25:
Currently, the collision models are only on the wheels. Ideally, the body of the vehicle so it won't be possible to drive through a concrete column as shown below:

[image]

jwnimmer-tri, 2017-01-13:
@rpoyner-tri I think perhaps a collision-box model here could be helpful?  I don't think the full body mesh would be. Any advice based on your other work?

SeanCurtis-TRI, 2017-01-13:
Would we want something in the middle?  Some convex shape that is a tighter fit than a box, but still an order of magnitude simpler than the visualization mesh?

liangfok, 2017-01-13:
That would be ideal. What app would you recommend using to create such a convex shape? Blender?

SeanCurtis-TRI, 2017-01-13:
Blender is one option; it's what I did for the iiwa.   I'm more proficient in Maya (attribute that to night on to twenty years of experience) and Thor has a license that may be briefly moochable.  Either way, it's maybe a 30 minute task.

jwnimmer-tri, 2017-01-13:
I'm not entirely sure what purpose that would serve.  We don't need to simulate body collisions, just acknowledge when first contact occurs.  A conservative box is likely fine for that.

liangfok, 2017-01-13:
Yes, a conservative box will likely be OK for most situations, and could be a first-spiral solution.