RobotLocomotion/drake #2461: No clear way to determine if a revision passed continuous

ggould-tri, 2016-05-26:
Our process requires in some cases that particular commits pass continuous (see, eg, the case of a fixing patch in the revert process).  However it is very difficult to determine when this has happened without clicking through to every continuous builder's page in Jenkins (also it seems to be entirely impossible via cdash).

In buildbot, for instance, there are time- and revision-centric interfaces; see e.g. http://trac.buildbot.net/wiki/ScreenShots .  Jenkins and CDash seem to lack such a view, or at least to lack in our configuration.

jamiesnape, 2016-05-27:
Best I can do is this:

https://drake-jenkins.csail.mit.edu/view/Continuous/job/linux-clang-continuous-debug/changes

Otherwise, it is out of our control for Jenkins.

jamiesnape, 2016-05-27:
I am asking about possibilities for CDash.

jamiesnape, 2016-06-28:
We can now filter CDash builds by revision, for example:

https://drake-cdash.csail.mit.edu/index.php?project=Drake&filtercount=1&showfilters=1&field1=revision&compare1=61&value1=d57f429a75347ea58f7792be34585d39bd5a1063

david-german-tri, 2016-11-04:
FYI, filtering CDash by revision always comes up empty for me.  Low priority, though.

jamiesnape, 2016-11-04:
Seems fine. The nightly revision was 48de280b26b451458bbd2127a4d14cb435b315f4. Hence,
https://drake-cdash.csail.mit.edu/index.php?project=Drake&filtercount=1&showfilters=1&field1=revision&compare1=63&value1=48de280b26b451458bbd2127a4d14cb435b315f4

david-german-tri, 2016-11-04:
Huh, OK, that link works.

closed 2016-11-04 by david-german-tri