RobotLocomotion/drake #2465: Document our code_style_guide rules for python

jwnimmer-tri, 2016-05-27:
We need to document our `code_style_guide.rst` rules for python code.  We have C++, Java, and MATLAB rules now, but not python.

For the record, in #1741 we choose googlestyle for C++ but didn't appear to discuss other languages.  I don't think there's anything else major we've missed, though.

jwnimmer-tri, 2016-05-27:
For starters, I think its obvious to elect [PEP-8](https://www.python.org/dev/peps/pep-0008/) as the basic rules.  We can tack on more beyond that if we start to care.

Probably also worth documenting how to run the `pep8` style-checker tool.

liangfok, 2016-05-27:
+1 for PEP-8.

jamiesnape, 2016-05-27:
Would you like me to run `flake8` or `pylint` on PR's.

jamiesnape, 2016-05-27:
The Google style guide for Python seems reasonable. As far as I can tell it is PEP8 plus a few other things.

jwnimmer-tri, 2016-05-27:
> Would you like me to run flake8 or pylint on PR's?

Possibly in the future, but for now I'd suggest that someone to test it locally and in integrate into drake-distro first, and then toss it into jenkins.   Also there's not _that_ much python code in Drake, so probably not a high priority.

david-german-tri, 2016-06-01:
Sounds reasonable to me.

rdeits, 2016-06-06:
FYI: the `pep8` package has been renamed to the more-readable [pycodestyle](https://pypi.python.org/pypi/pycodestyle) package going forward.

jamiesnape, 2016-06-06:
There is also [yapf](https://github.com/google/yapf) to autoformat Python files in the same spirit as `clang-format`.

jwnimmer-tri, 2016-07-06:
Let me try kicking this over to @david-german-tri, since I'm not foresee-ably going to work on it, ever.

david-german-tri, 2016-08-08:
I'm never going to work on this either.  Dropping it to the backlog.

jwnimmer-tri, 2016-11-30:
I wonder if @rpoyner-tri would be interested in taking a stab at this, and perhaps relating it to TRI python style?

rpoyner-tri, 2016-11-30:
Sure.

PR #4663 (2017-01-11): Begin adding Python coding style advice.
Here's a start for general review and discussion. PEP8 is the meat of it, though there are some additional rules suggested by folks working on internal TRI projects.

Closes #2465.

Of note:

 * I haven't made any attempt to make existing python files comply.
 * I wouldn't object to working on a detailed pylint configuration, but that feels like a separate PR.
 * My tool installation instructions put pylint within the user's account space -- no more `sudo pip install` nonsense. :)

---
This change is [[image]](https://reviewable.io/reviews/robotlocomoti [... cut]
closed 2017-01-11 by sammy-tri