RobotLocomotion/drake #2467: Does collision modeling require RigidBodySystem to have non-numerical state?

david-german-tri, 2016-05-27:
And if so, is that a bad thing?  Discussion needed.  See #2004.

sherm1, 2016-05-27:
> Does collision modeling require RigidBodySystem to have non-numerical state?

Yes. But not as we are using the term *state* in Drake, where it is restricted to `x`, a vector of numerical continuous and discrete variables. But there is plenty of non-numerical "mode" data (which is state information in the generic sense of the word). The contact manifold in #2004 is one example, the active set of constraints is another. There are plenty more and it is only a terminology question whether to call them "state" -- we've been using "Context" as a more inclusive term. A less-exotic example: min/max values are part of the context; they can't be calculated from the current set of states and modes in the Context.

> is that a bad thing?

Not at all!

david-german-tri, 2016-08-08:
(just returned here while skimming my assigned issues)

I think I'm missing a key intuition.  Let me come at this question another way.  As I understand it, [RigidBodySystem 1.0](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodySystem.h) has only numerical state.  Indeed, it has only continuous state: q and v.  Why does that work?

sherm1, 2016-08-08:
> Why does that work?

You are assuming it works! It has very limited functionality and does not provide discrete state at all (not to mention error control, caching, modes, parameters, guard(witness)/reset(event handler) functions, ...). The fully-functional Drake uses Matlab classes and the Simulink solver to achieve its ends, that's what we want System 2/Solver 2 to replace.

david-german-tri, 2016-08-09:
Yes, of course, I understand those limitations of System 1 in general.  However, the existing RigidBodySystem built on top of System1 uses only continuous state, and provides some kind of collision modeling.  Is there non-continuous information about the configuration of a real RigidBodyTree that this model fails to capture?

Put another way: from my limited grasp of the #2004 discussion, I cannot tell whether the proposed mode variables could be reconstructed from the continuous state or not.  If they can, then I wonder whether they are really `Cache` and not `State`.  If they cannot, it suggests there is an inherent tradeoff between fidelity of the model and the glass-box-ness of the model, since mode variables are essentially black-box.  Since RigidBodyTree is at the heart of Drake, the existence of such a fundamental tradeoff seems like it would be bad news.

sherm1, 2016-08-09:
> Is there non-continuous information about the configuration of a real, physical tree of rigid bodies that this model fails to capture?

This is a question about modeling, not about the physical world which is entirely continuous. RBSystem 1 supports only continuous (compliant) models of contact so there is no discrete information to capture. For many reasons, it is often preferable to approximate compliant contact with rigid models, which then introduces discrete behavior. In practice the difference is whether the contact is modeled using force elements (like springs and dampers) or with hard constraints. 

I agree with you completely that the distinction between state and cache is whether the quantity in question can be reconstructed from other quantities in the context. If so, it is cache. Mode variables should only hold values that cannot be reconstructed. For example, they must be used to implement a delay buffer, because that involves past information that is long gone and couldn't be reconstructed at the present time. 

Use of mode variables for rigid contact isn't as obvious, and some contact modeling methods don't require past information at all. Some that we want to use, however, require knowing the current active set (that is, which constraints are currently generating forces) in order to correctly isolate events and to compute what the active set is in the next time step. (The active set cannot be accurately reconstructed from the current state for several reasons including inexact tolerances.) I would be happy to discuss this live if you want to know more about it.

david-german-tri, 2016-12-13:
(cleaning up old issues)  This was useful, thanks @sherm1!  It sounds like the answer is yes, abstract state will be coming to RigidBodyPlant eventually.

closed 2016-12-13 by david-german-tri

RussTedrake, 2016-12-16:
@david-german-tri , @sherm1 -- fwiw, I'd like to be involved in this when it happens.  we want to be careful to avoid making rigid body systems impenetrable for synthesis/verification.  In matlab, we differentiated between hybridrigidbodymanipulators and rigidbodymanipulators.  i think having hybrid state in the main rigid body plant is better, but we just need to walk through a few use cases to make sure it's done thoughtfully.