RobotLocomotion/drake #2468: Document code_style_guide rules for CMake

jamiesnape, 2016-05-27:
It is all pretty random at present. I would suggest keeping it short, though. Maybe the following:

---

CMake Style
===========

We follow [KDE CMake Coding Style](https://community.kde.org/Policies/CMake_Coding_Style) and the additional rules noted below.

* Do not add a space between a function or macro call and the opening
  parenthesis. This includes the opening of blocks such as `foreach()`, `if()`, and `while()`.
* It is recommended, but not required, to wrap lines at 80 characters  _(or 120 or not at all?)_ . Indent continued lines by two extra spaces _(or four?)_. This includes arguments to functions and macros.
* Use all-lowercase style (e.g., `add_executable()` instead of `ADD_EXECUTABLE()`) except where the [CMake Documentation](https://cmake.org/cmake/help/latest/) uses otherwise (e.g., `ExternalProject_Add()`).

---

This matches a plurality of code except for `if()` vs. `if ()`, which dominates for inexplicable reasons, and the line wrapping.

There is a [cmakelint](https://github.com/richq/cmake-lint) program we could selectively enable. Not sure of its quality, but may be something to build on.

jwnimmer-tri, 2018-11-14:
@jamiesnape Do you think this is still worth tracking?  Mostly only you write CMake, I think?  Not sure its worth having a de-jure style guide (for feature and platform reviewers to learn and enforce), when we use it so little?

jamiesnape, 2018-11-14:
Probably not.

closed 2018-11-14 by jamiesnape