RobotLocomotion/drake #2478: Building with clang fails with -Wself-assign

rdeits, 2016-05-31:
On master, with clang 703 (OSX):

```
[ 97%] Building CXX object bindings/swig/CMakeFiles/_ikPython.dir/ikPYTHON_wrap.cxx.o
/Users/rdeits/locomotion/drake-distro/drake/pod-build/bindings/swig/ikPYTHON_wrap.cxx:2427:23: error: explicitly assigning value of variable of type 'int' to itself [-Werror,-Wself-assign]
                  res = (res);
                  ~~~ ^  ~~~
/Users/rdeits/locomotion/drake-distro/drake/pod-build/bindings/swig/ikPYTHON_wrap.cxx:2430:23: error: explicitly assigning value of variable of type 'int' to itself [-Werror,-Wself-assign]
                  res = (res);
                  ~~~ ^  ~~~
2 errors generated.
make[6]: *** [bindings/swig/CMakeFiles/_ikPython.dir/ikPYTHON_wrap.cxx.o] Error 1
make[5]: *** [bindings/swig/CMakeFiles/_ikPython.dir/all] Error 2
make[4]: *** [all] Error 2
```

This causes the build to fail because we have `-Werror` set. We can squash the warning with `-Wno-self-assign`, but there's probably a better way.

jwnimmer-tri, 2016-05-31:
Interesting, `-Wself-assign` is part of `-Wall` so our `-Werror=all` must be tickling that.  I would be super-interested to learn why Jenkins isn't catching this, not even in the full Continuous builder.  I think that's way more troubling than this particular instance of this particular warning.

In general, fixing warnings in swig-generated code may be a losing battle, so using `-Werror=...` within that directory may not even be worth attempting.  In my ea11c1f work in progress on `-Wsign-compare`, I've decided not to fix it in the swig code.  Turning off errors in `_wrap` files would be +1'd by me.

david-german-tri, 2016-05-31:
I agree with @jwnimmer-tri on all counts.  I'll take a look into why Jenkins isn't catching the failure.

david-german-tri, 2016-05-31:
Also, I can reproduce the issue on my local Mac.

david-german-tri, 2016-05-31:
Here's an example compilation step from `make` that fails locally for me:

```
[ 98%] Building CXX object bindings/swig/CMakeFiles/_rbtPython.dir/rbtreePYTHON_wrap.cxx.o
cd /Users/davidgerman/drake-distro/drake/pod-build/bindings/swig && /usr/local/opt/ccache/libexec/c++   -D_rbtPython_EXPORTS -I/Users/davidgerman/drake-distro/drake/pod-build/generated -I/Users/davidgerman/drake-distro/drake/.. -I/Users/davidgerman/drake-distro/drake/pod-build/exports -isystem /Users/davidgerman/drake-distro/build/include -I/Users/davidgerman/drake-distro/drake/pod-build/lcmgen -isystem /usr/local/Cellar/glib/2.46.2/include/glib-2.0 -isystem /usr/local/Cellar/glib/2.46.2/lib/glib-2.0/include -isystem /usr/local/opt/gettext/include -isystem /Users/davidgerman/drake-distro/build/include/eigen3 -I/Users/davidgerman/drake-distro/drake/thirdParty/spruce/include -I/Users/davidgerman/drake-distro/drake/thirdParty/cimg -I/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/usr/local/lib/python2.7/site-packages/numpy/core/include -I/Applications/MATLAB_R2016a.app/extern/include -I/Applications/MATLAB_R2016a.app/simulink/include  -Werror=all -Wno-sign-compare -Wno-missing-braces -Wno-unused-private-field -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -fPIC   -Wno-parentheses-equality -std=gnu++11 -o CMakeFiles/_rbtPython.dir/rbtreePYTHON_wrap.cxx.o -c /Users/davidgerman/drake-distro/drake/pod-build/bindings/swig/rbtreePYTHON_wrap.cxx
```

Here's an analogous step that succeeds on Jenkins, and also succeeds locally (after fixing up the paths):

```
10:24:52 [ 83%] Building CXX object bindings/swig/CMakeFiles/_rbtPython.dir/rbtreePYTHON_wrap.cxx.o
10:24:52 cd /Users/yosemite/workspace/experimental/5403bcc7/drake/pod-build/bindings/swig && "/usr/local/Cellar/cmake/3.5.2/bin/ctest" --launch --target-name _rbtPython --build-dir /Users/yosemite/workspace/experimental/5403bcc7/drake/pod-build/bindings/swig --output CMakeFiles/_rbtPython.dir/rbtreePYTHON_wrap.cxx.o --source /Users/yosemite/workspace/experimental/5403bcc7/drake/pod-build/bindings/swig/rbtreePYTHON_wrap.cxx --language CXX -- /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++   -D_rbtPython_EXPORTS -I/Users/yosemite/workspace/experimental/5403bcc7/drake/pod-build/generated -I/Users/yosemite/workspace/experimental/5403bcc7/drake/.. -I/Users/yosemite/workspace/experimental/5403bcc7/drake/pod-build/exports -isystem /Users/yosemite/workspace/experimental/5403bcc7/build/include -I/Users/yosemite/workspace/experimental/5403bcc7/drake/pod-build/lcmgen -isystem /usr/local/Cellar/glib/2.46.2/include/glib-2.0 -isystem /usr/local/Cellar/glib/2.46.2/lib/glib-2.0/include -isystem /usr/local/opt/gettext/include -isystem /Users/yosemite/workspace/experimental/5403bcc7/build/include/eigen3 -I/Users/yosemite/workspace/experimental/5403bcc7/drake/thirdParty/spruce/include -I/Users/yosemite/workspace/experimental/5403bcc7/drake/thirdParty/cimg -I/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/usr/local/lib/python2.7/site-packages/numpy/core/include  -Werror=all -Wno-sign-compare -Wno-missing-braces -Wno-unused-private-field -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -O3 -DNDEBUG -fPIC   -Wno-parentheses-equality -std=gnu++11 -o CMakeFiles/_rbtPython.dir/rbtreePYTHON_wrap.cxx.o -c /Users/yosemite/workspace/experimental/5403bcc7/drake/pod-build/bindings/swig/rbtreePYTHON_wrap.cxx
```

david-german-tri, 2016-05-31:
@jamiesnape It looks like `ctest` might not be passing `-Werror=all` through to clang.  Any ideas?

jamiesnape, 2016-05-31:
@david-german-tri I am seeing it in all the output in the Jenkins build you linked. Try removing your local pod-build directories and recompile.

jamiesnape, 2016-05-31:
@rdeits Build with verbose output and re-post the error. We need all the compiler flags that were used.

david-german-tri, 2016-05-31:
Oh, I agree the flag is on the command line, it just doesn't seem to be having an effect!

jamiesnape, 2016-05-31:
If you compile directly with `clang` instead of `ccache` does it fail?

jamiesnape, 2016-05-31:
Also CI runs OS X 10.10, which is different to your build. There is not official vSphere support for 10.11 yet, so 10.10 is as recent as we can do.

david-german-tri, 2016-05-31:
Huh, I am glad you asked.  When I replace `ccache` with `clang++`, the compile succeeds.  That's baffling.

```
cd /Users/davidgerman/drake-distro/drake/pod-build/bindings/swig && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++   -D_rbtPython_EXPORTS -I/Users/davidgerman/drake-distro/drake/pod-build/generated -I/Users/davidgerman/drake-distro/drake/.. -I/Users/davidgerman/drake-distro/drake/pod-build/exports -isystem /Users/davidgerman/drake-distro/build/include -I/Users/davidgerman/drake-distro/drake/pod-build/lcmgen -isystem /usr/local/Cellar/glib/2.46.2/include/glib-2.0 -isystem /usr/local/Cellar/glib/2.46.2/lib/glib-2.0/include -isystem /usr/local/opt/gettext/include -isystem /Users/davidgerman/drake-distro/build/include/eigen3 -I/Users/davidgerman/drake-distro/drake/thirdParty/spruce/include -I/Users/davidgerman/drake-distro/drake/thirdParty/cimg -I/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/usr/local/lib/python2.7/site-packages/numpy/core/include -I/Applications/MATLAB_R2016a.app/extern/include -I/Applications/MATLAB_R2016a.app/simulink/include  -Werror=all -Wno-sign-compare -Wno-missing-braces -Wno-unused-private-field -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -fPIC   -Wno-parentheses-equality -std=gnu++11 -o CMakeFiles/_rbtPython.dir/rbtreePYTHON_wrap.cxx.o -c /Users/davidgerman/drake-distro/drake/pod-build/bindings/swig/rbtreePYTHON_wrap.cxx
```

jamiesnape, 2016-05-31:
I do see one possible bug. The ID of the Apple compiler changed to `AppleClang` while back.

Output of CMake for this project is `Clang`.
```
project(Test)
if(CMAKE_CXX_COMPILER_ID STREQUAL "Clang")
  message("Clang")
else()
  message("Not Clang")
endif()
```

Output of CMake for this project is `Not Clang`.
```
cmake_minimum_required(VERSION 3.5)  # ADDED
project(Test)
if(CMAKE_CXX_COMPILER_ID STREQUAL "Clang")
  message("Clang")
else()
  message("Not Clang")
endif()
```

Output of CMake for this project is `Clang`.
```
cmake_minimum_required(VERSION 3.5)
project(Test)
if(CMAKE_CXX_COMPILER_ID MATCHES "Clang")  # CHANGED
  message("Clang")
else()
  message("Not Clang")
endif()
```

david-german-tri, 2016-05-31:
Aha: http://petereisentraut.blogspot.com/2011/09/ccache-and-clang-part-2.html

@rdeits, are you also using `ccache`?

jamiesnape, 2016-05-31:
You should definitely set `export CCACHE_CPP2=yes`.

david-german-tri, 2016-05-31:
I'm going to go ahead and close this.  @rdeits, feel free to reopen if needed.

jwnimmer-tri, 2016-05-31:
We document in sphinx how to turn on `ccache`; shouldn't we also document the mandatory environment variable for it when used with clang, too?

jamiesnape, 2016-05-31:
Yes.

rdeits, 2016-06-01:
Yes, setting CCACHE_CPP2 fixes it for me.

PR #2511 (2016-06-07): Remove ccache from the Drake documentation.
We shouldn't maintain documentation for an optional convenience tool that we don't test in CI.

Fixes #2478.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2511)
closed 2016-06-07 by david-german-tri