RobotLocomotion/drake #2484: Remove all manually created makefiles

jamiesnape, 2016-05-31:
All makefiles should be generated by CMake.

david-german-tri, 2016-06-08:
This ticket refers to the top-level Makefile, and to custom externals / CMake PODS.

jwnimmer-tri, 2016-07-13:
@RussTedrake Let's discuss PODS here instead of in PR #2758.

We have a goal in #2482 of not relying on PODS within Drake to build our externals -- to use CMake exclusively to compile everything within `drake-distro` (to the extent possible).  That seems fine.

The question is, should we still provide a `drake-distro/Makefile` in our source tree, for projects that want to use Drake as one of their externals -- i.e., a simple Makefile with just the `all` target that shells out to CMake to do the build + install into `BUILD_PREFIX`.

Does anyone still need or use that capability?

wxmerkt, 2016-07-13:
OpenHumanoids needs the ``drake-distro/drake/Makefile``

jamiesnape, 2016-07-13:
@wxmerkt Would a particular target generated by CMake work or do you physically have to have a `Makefile` at that path?

wxmerkt, 2016-07-13:
@jamiesnape Drake's being called from a top-level PODS build system so we do need to have a Makefile at that path.

mwoehlke-kitware, 2016-07-13:
> OpenHumanoids needs the `drake-distro/drake/Makefile` (as we provide required externals ourselves)

The file you reference is https://github.com/RobotLocomotion/drake/blob/master/drake/Makefile, correct? I take this to mean you are building *just* drake, and not the drake superbuild? Do you need [`<drake>/Makefile`](https://github.com/RobotLocomotion/drake/blob/master/Makefile)?

wxmerkt, 2016-07-13:
@mwoehlke-kitware Correct, we do not need ``drake/Makefile``, only ``drake/drake/Makefile``. I've seen Jeremy's comment only referencing the superbuild-Makefile, but commented given the title of the issue to remove _all_.

jwnimmer-tri, 2016-07-13:
Thanks, that's very helpful to know.

david-german-tri, 2016-07-14:
OK, I'm going to create a PR that zaps the top-level Makefile, since it appears that its only purpose in life is now to confuse people.

mwoehlke-kitware, 2016-07-14:
@david-german-tri: #2809

mwoehlke-kitware, 2016-09-08:
Isn't this dealt with to the extent possible? Last I knew, folks were insistent that we keep the remaining `Makefile`...

david-german-tri, 2016-09-08:
Agreed.

closed 2016-09-08 by david-german-tri