RobotLocomotion/drake #2486: Move LCM consumers to CMake-ified LCM

mwoehlke-kitware, 2016-05-31:
There is now a usable version of CMake-ified LCM (https://github.com/RobotLocomotion/lcm/tree/cmake). We should port all users of LCM to this, so that we can kill off the PODS lcm functions (i.e. `lcm.cmake` in https://github.com/RobotLocomotion/cmake/). This will bring us closer to not needing PODS or the aforementioned "cmake" repository.

mwoehlke-kitware, 2016-05-31:
Done for bot_core_lcmtypes: openhumanoids/bot_core_lcmtypes#18

Since this is going to cause a ripple effect with downstream consumers needing to switch as their upstream dependencies switch, we likely won't touch drake itself until last, so we can update everyone at once and have a working build.

david-german-tri, 2016-06-08:
Upgrading this to `priority: high` because it blocks #2341.  Any news?

mwoehlke-kitware, 2016-06-08:
I have this locally working, but it's based on #2487; I'm hoping that will land first so I don't have to go back and redo the changes to externals (which would then conflict with #2487). However, see openhumanoids/bot_core_lcmtypes#18; upstream is rejecting the PR, so unless that can be resolved, we would have to use our own copy of bot_core_lcmtypes.

Also, I should note for the record that #2483 depends on this.

sammy-tri, 2016-09-13:
FWIW, it's been somewhat difficult for me to figure out the right way to (for example) reference lcm types defined in bot_core_lcmtypes from within Drake.  I believe this is because the way the output is created/built from the .lcm files is different.  I suspect that moving Drake to using the same lcm cmake as the other libraries would simplify the matter.

sammy-tri, 2016-09-13:
Actually I think what I'm describing above is really just #3351.

mwoehlke-kitware, 2016-09-13:
>  I suspect that moving Drake to using the same lcm cmake as the other libraries would simplify the matter.

It already does. Am I missing something?

> Actually I think what I'm describing above is really just #3351.

Maybe :smile:. BTW,  please feel free to "vote" on if `lcmtypes/` should be part of how we name the includes... (If there is no strong consensus, omitting will likely win by default as that option does not require changes to upstream LCM.)

mwoehlke-kitware, 2016-09-13:
Only loosely related, but this can probably be closed. There is an ongoing task to figure out what to do with bot_core_lcmtypes, but otherwise I think this is effectively "done". Any objections?

sammy-tri, 2016-09-13:
Welp, I'm more confused than ever (but pretty sure I am hitting #3351, where I have no preference for the outcome as long as it's consistent).

In terms of how lcm types are built, I see at least two (maybe three) patterns..

bot_core_lcmtypes seems to do include(${LCM_USE_FILE}) which pulls in the cmake lcm macros.  Drake does this too, but it also has drake/cmake/lcmtypes.cmake -- is that file now unused?

libbot appears to include 5 identical copies of lcmtypes.cmake, which are different than the one in drake/cmake.

This comment can probably be ignored if nobody else has any more of an idea what I'm talking about than I do (not much!).  I should probably not have chimed in since I was confusing this with another ticket anyway.  Sigh.

mwoehlke-kitware, 2016-09-13:
> Drake [...] has `drake/cmake/lcmtypes.cmake` -- is that file now unused?

Yes. Note that that is actually [RobotLocomotion/cmake:lcmtypes.cmake](https://github.com/RobotLocomotion/cmake/blob/master/lcmtypes.cmake), so just getting rid of it is not straight forward. (Hopefully in the not so distant future, however, we will be able to drop RobotLocomotion/cmake entirely.)

> libbot appears to include 5 identical copies of lcmtypes.cmake

libbot, at least the monolithic repository, is effectively unmaintained. It was never updated and almost certainly never will be. (But see previous comment about the future of bot_core_lcmtypes; some or all of this may move back into the libbot organization as a new repository.)

david-german-tri, 2016-10-06:
Might be done; @mwoehlke-kitware to confirm.

mwoehlke-kitware, 2016-10-06:
> Might be done; @mwoehlke-kitware to confirm.

Heh... hadn't looked in a while, now that I'm reading the old comments...

> Only loosely related, but this can probably be closed. There is an ongoing task to figure out what to do with bot_core_lcmtypes, but otherwise I think this is effectively "done". Any objections?

...so I think it's "Executive Decision" time :smile:.

(Note: LCM types are now [RobotLocomotion/lcmtypes](https://github.com/RobotLocomotion/lcmtypes), so that's been sorted.)

closed 2016-10-06 by mwoehlke-kitware