RobotLocomotion/drake #2494: install on fresh mac failed to find vtk5

RussTedrake, 2016-06-01:
```
russt$ brew install vtk5 --with-qt
Error: No available formula with the name "vtk5"
```

jamiesnape, 2016-06-01:
The formula no longer exists, see https://github.com/Homebrew/homebrew-science/pull/3616. We will need to find an alternative or switch to a supported version of VTK.

FYI @billhoffman.

jamiesnape, 2016-06-01:
```
$ brew search vtk5
No formula found for "vtk5".

$ brew info vtk
homebrew/science/vtk: stable 7.0.0 (bottled), HEAD
Toolkit for 3D computer graphics, image processing, and visualization.
http://www.vtk.org
Not installed
From: https://github.com/Homebrew/homebrew-science/blob/master/vtk.rb
==> Dependencies
Build: cmake ✔
Recommended: boost ✔, fontconfig ✔, hdf5 ✔, jpeg ✔, libpng ✔, libtiff ✔
Optional: qt ✔, qt5 ✘
==> Options
--c++11
	Build using C++11 mode
--with-examples
	Compile and install various examples
--with-matplotlib
	Enable matplotlib support
--with-python3
	Build with python3 support
--with-qt
	Build with qt support
--with-qt-extern
	Enable Qt4 extension via non-Homebrew external Qt4
--with-qt5
	Build with qt5 support
--with-tcl
	Enable Tcl wrapping of VTK classes
--with-x11
	Build with x11 support
--without-boost
	Build without boost support
--without-fontconfig
	Build without fontconfig support
--without-hdf5
	Build without hdf5 support
--without-jpeg
	Build without jpeg support
--without-legacy
	Disable legacy APIs
--without-libpng
	Build without libpng support
--without-libtiff
	Build without libtiff support
--without-python
	Build without python2 support
--HEAD
	Install HEAD version
==> Caveats
Even without the --with-qt option, you can display native VTK render windows
from python. Alternatively, you can integrate the RenderWindowInteractor
in PyQt, PySide, Tk or Wx at runtime. Read more:
    import vtk.qt4; help(vtk.qt4) or import vtk.wx; help(vtk.wx)
```

rdeits, 2016-06-01:
I was just in the process of looking into this myself. 

Another option is that we could host our own Homebrew tap with the vtk5 formula. To do that, we would just need to create a new repo like `RobotLocomotion/homebrew` and then put the `vtk5.rb` file in that repo. Then users could install vtk5 with

`brew install RobotLocomotion/homebrew/vtk5`

rdeits, 2016-06-01:
See: https://github.com/Homebrew/brew/blob/master/share/doc/homebrew/How-to-Create-and-Maintain-a-Tap.md

jamiesnape, 2016-06-01:
We could, but I would vote strongly against it.

jamiesnape, 2016-06-01:
@RussTedrake Could you remind we why we are using VTK 5? Is upgrading to VTK 7 something within our control?

rdeits, 2016-06-01:
VTK5 is used within the Director, and since @patmarion is off all summer I think we're probably going to need it for the next few months at least. 

I set up a tap on my own account for testing, so I believe it should be possible to install vtk5 with:

```
brew tap rdeits/director
brew install vtk5
```

or 

```
brew install rdeits/director/vtk5
```

jamiesnape, 2016-06-01:
Ok. We can do that for now. @RussTedrake Could you create `RobotLocomotion/homebrew-director` and give me push access?

RussTedrake, 2016-06-01:
done

PR #2512 (2016-06-07): Update Homebrew installation instructions with new tap for vtk5
Fixes #2494.

FYI @RussTedrake.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2512)
commit fd81f464 (2016-06-07): Update Homebrew installation instructions with new tap for vtk5

Fixes #2494.
closed 2016-06-07 by jamiesnape