RobotLocomotion/drake #2501: Tracking issue for Windows-specific development cost

david-german-tri, 2016-06-02:
This issue is for gathering data on PRs that incur Windows-specific development cost.  It is not a forum for discussing the implications.  Examples of "cost" include:

* Developer effort spent resolving Windows-specific issues.
* Delayed merge due to discussing or resolving Windows-specific issues.
* Windows-specific flakes in pre-merge continuous integration.
* Rollbacks due to failures in Windows post-merge continuous integration.

I plan to reference this issue from PRs using the following boilerplate:

> #2499: two business days of latency, 540 minutes of developer effort

liangfok, 2016-06-02:
https://github.com/RobotLocomotion/drake/pull/2171: 9 days of latency, 180 minutes of developer effort

billhoffman, 2016-06-09:
https://github.com/RobotLocomotion/drake/pull/2171 seems more related to 32 bit vs 64 bit.  Since Windows is the only place that 32 bit is being tested it showed up there.  I would not classify this as a "Windows" issue.

liangfok, 2016-06-28:
https://github.com/RobotLocomotion/drake/pull/2679: 30 minutes of developer effort (maybe this is a 32-bit vs. 64 bit issue and thus shoulnd't be classified as a "Windows" issue as @billhoffman notes in previous comment).

jamiesnape, 2016-08-03:
I now have a week of CI spot data. We spend 84% on Windows instances, 16% on Linux.

ggould-tri, 2016-08-03:
I just spent most of two days on #2554 -- one day setting and debugging a windows environment, the second trying to suss out cmake and pkg-config behaviour on windows (through a soda-straw rdesktop connection).  Ultimately failed.

david-german-tri, 2016-08-17:
#3135: 60 minutes resolving MSVC-specific compile errors C2306 (forward declarations don't suffice for template instantiation) and C2248 (copy constructor can't reference private members).

liangfok, 2016-08-19:
The following msvc-ninja-32-windows-debug CI test took 57 minutes to complete:

https://drake-jenkins.csail.mit.edu/job/experimental/3875/compiler=msvc-ninja-32,label=windows,type=debug/

[image]

liangfok, 2016-10-04:
FTR, #3643 was delayed about 24 hours due to a mysterious Win32 linker error. Fixing it required adding something to the CMakeLists.txt: https://github.com/RobotLocomotion/drake/pull/3643/commits/6def3466e74da3bf617feb4a7ebefc2c31bcd37f#diff-7a7595a8eccbcad1fb24437727b22ebcR34 and https://github.com/RobotLocomotion/drake/pull/3643/commits/6def3466e74da3bf617feb4a7ebefc2c31bcd37f#diff-7a7595a8eccbcad1fb24437727b22ebcR52.

david-german-tri, 2016-10-06:
Windows is no longer supported. :tada:

closed 2016-10-06 by david-german-tri