RobotLocomotion/drake #2508: removing java -source 6 broke drake.jar on matlab 2015b for OSX

RussTedrake, 2016-06-05:
reverting https://github.com/RobotLocomotion/drake/pull/2404 resolves it for me.  
(cc @david-german-tri)

The specific failure for me was that I could not communicate via lcm to the BotVisualizer.  The build servers do not have adequate coverage of this case.  We should consider removing the lines about visualizers from here:
https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L189

rdeits, 2016-06-06:
I think this is the same issue I reported in #2473 (also on OSX)

rdeits, 2016-06-06:
Also, maybe we can finally close #1256 😉

david-german-tri, 2016-06-06:
I agree with @rdeits that this sounds like a duplicate of #2473.  Since we already have a bunch of notes on that issue, I'll close this one.

closed 2016-06-06 by david-german-tri