RobotLocomotion/drake #2514: Clarify "Integer types" rule from cppguide

jwnimmer-tri, 2016-06-07:
https://google.github.io/styleguide/cppguide.html#Integer_Types

We should clarify the "Integer types" rule from cppguide.  It's not clear when (other than bitfields) unsigned types can be used.

The rule that david-german-tri and I seem to agree on is that unsigned integers may only be used:
- for bitfields
- when a (1) non-Drake API uses unsigned types, and (2) casting to a signed type during Drake's interactions with the non-Drake API would obscure the readability of our code, and (3) subtraction underflow is obviously not at risk. 

Under those three conditions, `size_t` (and not any other unsigned type) may be used by Drake.

jwnimmer-tri, 2016-06-07:
Possible additional item to fix at the same time: use int64_t, not ptrdiff_t, within Drake.  By having fewer choices of integer type (`int`, `int64_t`, and a few `size_t`), code will be easier to read and write.

sherm1, 2016-06-07:
`int64_t` seems too specific in many cases. It is rare that we require exactly 64 bits, but rather want the biggest integer available. I like `long long` for that purpose. It has several advantages:
- it is available on all conformant compilers (int64_t doesn't have to be there, although it probably will be)
- there is no issue about whether we have to specify `std::` first (we'll need a style exception to avoid `std::int64_t`)
- it isn't being overly precise about the number of bits that we don't actually care about for loop counters and element indices.

But, didn't we decide on `ptrdiff_t` because Eigen uses it? This is from the Eigen 3.3 docs:
```
Index typedef
In Eigen 3.3, the "Index" typedef is now global and defined by default to std::ptrdiff_t:

namespace Eigen {
  typedef std::ptrdiff_t Index;
}
```

sammy-tri, 2016-06-07:
Thoughts:

* I think if you're actually doing arithmetic on bare pointers (which is likely to be difficult to explain why this is a good idea during review anyway), ptrdiff_t is strictly correct and should be used in that case.  Using it as an arbitrary index variable type seems less obviously correct, and if a third party library is doing it than the spelling "Eigen::Index" seems to be the right way to go.

* In its text, the google style guide never prepends std:: to the C99 integer types, and none of our supported compilers seem to need it anyway.  I'd say it doesn't need an exception in Drake.

sherm1, 2016-06-07:
> I'd say it doesn't need an exception in Drake.

We already have an "additional rule" covering this for size_t and ptrdiff_t [here](http://drake.mit.edu/code_style_guide.html#additional-rules) (3rd bullet), we would need to add `int64_t`. The rule is there to *prevent* people from cluttering up the code with `std::`, not to *permit* leaving it off.

sammy-tri, 2016-06-08:
I somehow missed that (which is weird, since I searched for std::).  Is there a reason to enumerate the integer types one should not use std:: with?  As opposed to "don't prepend std:: to anything from stdint or stddef" or "don't put std:: before integer types?"  (an idea which had honestly never occurred to me to begin with)

sherm1, 2016-06-09:
> Is there a reason to enumerate the integer types one should not use `std::` with? 

If we're only going to allow three or four integer types it is probably good to list them explictly. I'm not sure what else is in `<cstdint>` and `<cstddef>` and whether we would want to leave off `std::` for those. They would certainly be less common so the clutter is likely to be insignificant. I would propose we list the small number of currently-bothersome integer types and wait until we see a clutter problem before opening this up further.

jwnimmer-tri, 2016-06-28:
So #2677 is prepwork for this.  I'll have new text for this item in progress shortly.

jwnimmer-tri, 2016-06-28:
I'll add a rationale section here, for the record:

(1) Why disallow unsigned in general?  Same reason as cppguide use to have (but they removed the text) -- underflow below zero is a pothole waiting to be stepped in.

(2) Why unsigned bitfields?  They are not numbers, and we don't want sign-extension messiness.  We require fixed-size for bitsets, because the bitset extent should be well-defined by our code, and never vary by platform.  Allowing sets smaller than 32 bits adds no value -- just more complexity.

(3) Why allow `size_t`?  It's so prevalent in other APIs (looking at you `std::vector`) that sometimes you just have to go with the flow.

jwnimmer-tri, 2016-06-28:
And now on to the signed types.  I would argue the following...

In general, we should encourage (and probably allow) only two signed integer types in Drake:
- `int`
- `int64_t`

--

Per https://google.github.io/styleguide/cppguide.html#Integer_Types, `int` is at least 32 bits.  It will serve well for most purposes and should be the default choice.

> int64_t seems too specific in many cases. It is rare that we require exactly 64 bits, but rather want the biggest integer available. I like long long for that purpose.

Most Drake code should **not** "flex" to allow for different capabilities depending on the platform / compiler you have chosen.  If the thing you're counting or doing math on with will only ever need 32 bits, then `int` works great.  If you might need more than that, I don't want to guess and hope that some weird integer type will be long enough -- I want it to always work, so we should use `int64_t` in that case.

Note that this is different than Eigen, which tries to be a library that suits a very wide range of needs, and has magically-shrinking array indices depending on how you've compiled it.  I think that is bad example for Drake to use.

> Didn't we decide on ptrdiff_t because Eigen uses it?

Possibly, yes.  I am pushing back against that conclusion.

> I think if you're actually doing arithmetic on bare pointers (which is likely to be difficult to explain why this is a good idea during review anyway), ptrdiff_t is strictly correct and should be used in that case.

That may be fine, but I'd rather leave that out of the style guide.  We can litigate it only when it arises.

> Using it as an arbitrary index variable type seems less obviously correct, and if a third party library is doing it than the spelling "Eigen::Index" seems to be the right way to go.

For perhaps some generic Eigen-using code like some ADL helpers or other things that are essentially appendages to Eigen (could move into Eigen itself), using `Eigen::Index` might make sense.  For Drake though, I really think `int` and `int64_t` are superior.

--

One exception to the above is if you are representing a disk or network structure in C++ code, in which case you should use the entirety of  `<cstdint>` to you heart's content.

maddog-tri, 2016-06-28:
Two cents:  sounds like this is converging on blessing three types:  `size_t`, `int`, `int64_t`.  The guidelines should include some guidance on casting conventions.  E.g., if a `vector::size()` result is going to be stored in a signed type, should it be "a) always `int64_t`, where it is most likely to fit" or "b) possibly `int`, if the author is convinced that overflow will not be an issue"?

Is there any downside to comparing an `int` or `int64_t` against `size_t`?  E.g., `i < vector::size()` in a loop.

(In my recent experiences with mixing signed/unsigned types in an interface, though... mixing was just a means to use `<0` as an "invalid value" flag --- which would have been better served by the likes of `boost::optional` if it were available.)

sherm1, 2016-06-28:
I'm going to make a final argument to permit `long long` and then acquiesce to `int64_t` if y'all are so enamored with it. 

**(1)** note that the Google style guide mixes the two [here](https://google.github.io/styleguide/cppguide.html#64-bit_Portability):

> Use the LL or ULL suffixes as needed to create 64-bit constants. For example:
  int64_t my_value = 0x123456789LL;
  uint64_t my_mask = 3ULL << 48;

They have to do that since the language cleanly supports `long long` constants but not `int64_t` constants.

**(2)** the C++11 standard provides the following:
> long long - target type will have width of at least 64 bits.

That is a considerably better guarantee that what you get for `int`:
> int - guaranteed to have a width of at least 16 bits. 

(Summarized [here](http://en.cppreference.com/w/cpp/language/types).)

So with `long long` we get:
- built-in constants of matching type
- guarantee of lots of space
- no implication that our algorithms depend on exactly 64 bits
- guarantee of support on all platforms (int64_t is there in practice but not by promise)

This makes me think that the primary argument for using `int64_t` rather than `long long` (when you don't actually need exactly 64 bits) is that the name is shorter!

My proposal would be:
- use `int` wherever possible, and limit Drake support to machines where `int` is at least 32 bits
- use `long long` whenever you need more space than an int but don't need an exact number of bits; you'll get at least 64 bits worth
- use `int32_t`, `int64_t` and so on when you actually want to emphasize the bit count
- never use `long`
- use `ptrdiff_t` and `size_t` when forced to by someone else's API.

I would also be open to declaring that Drake is supported only on machines where an int is exactly 32 bits and a long long is exactly 64 bits.

jwnimmer-tri, 2016-06-29:
FWIW, I just did a grep of `drake-distro/drake` to see how and where `int64_t`, `long long`, and `ptrdiff_t` are used now:

- There are uses in `nice_type_name`, but that's just to be able to print things, not an actual desire for wide integers.
- There are lots of `int64_t` used in LCM messages and related code, but that falls under the "use fixed-sized integers for messages" exception.
- There is exactly 1 remaining `long long` and 1 `int64_t` use in the tree, in each case used for pointer-integer type punning in mex junk.  This is known to be awful and should not inform our nominal style choices.
- There are a handful of `ptrdiff_t` that were recent introductions, from some early attempts to answer this question, and which were what caused me to want to look at this rule in the first place.

As such, perhaps we could conclude that nobody actually _needs_ 64-bit-minimum integers right now, and so we punt on writing rules about them.  For now, I would be satisfied to just outlaw `ptrdiff_t` and then close this issue.

jwnimmer-tri, 2016-07-06:
With those two PRs in, I think this issue is solved well enough for now.  If we ever need a 64-bit integral type, we can revisit some of the comments above to seed the discussion.

closed 2016-07-06 by jwnimmer-tri