RobotLocomotion/drake #2516: Model Joint Position Limits Are Not Enforced

liangfok, 2016-06-08:
I wrote a [simple dynamics simulation for the KUKA IIWA arm](https://github.com/liangfok/drake/blob/add_kuka2/drake/examples/kuka_iiwa_arm/run_kuka_iiwa_arm_dynamics.cc) and included a check to verify that the final joint positions fall within the specified joint position limits. Unfortunately, the check reliably detects joint position limit violations.

To replicate the test, checkout branch `add_kuka2` of [my fork](https://github.com/liangfok/drake/tree/add_kuka2), then after compiling:

```
$ cd [drake build artifacts directory]
$ ctest -VV -R run_kuka_iiwa_arm_dynamics
```

Here is an example failure:

```
357: libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: ERROR: Joint iiwa_joint_2 (DOF iiwa_joint_2) violated minimum position limit (-2.111081 < -2.094395).
```

liangfok, 2016-06-08:
The forces for enforcing the joint limits are defined in [RigidBodySystem::dynamics()](https://github.com/RobotLocomotion/drake/blob/b4d3a9cd9f9acba0f1e29adb52ff6177fb587acb/drake/systems/plants/RigidBodySystem.cpp#L118). These forces are generated by a PD controller where `k = RigidBodySystem::penetration_stiffness` and `d = RigidBodySystem::penetration_damping`. Thus, I wonder if the degree to which joint limits are violated could be reduced by increasing these gains, and whether it is even possible to achieve zero joint limit violations given the fact that they are enforced by asserting a force that's proportional to the degree to which the joint position limits are violated.

RigidBodyTree also has joint limit vectors (`joint_limit_min` and `joint_limit_max`) that are declared [here](https://github.com/RobotLocomotion/drake/blob/b1a10841d642c70bf824043f57286d60de4272d8/drake/systems/plants/RigidBodyTree.h#L818) and initialized [here](https://github.com/RobotLocomotion/drake/blob/b1a10841d642c70bf824043f57286d60de4272d8/drake/systems/plants/RigidBodyTree.cpp#L174). These vectors are used by `RigidBodyTree::jointLimitConstraints()` and `RigidBodyTree::getNumJointLimitConstraints()`.

liangfok, 2016-06-08:
I hacked higher gains (`Kp = 5000` and `Kd = 100` versus the original `Kp = 3000` and `Kd = 0`) and the degree to which the joint position limits were violated was reduced:

> 357: libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: ERROR: Joint iiwa_joint_2 (DOF iiwa_joint_2) violated minimum position limit (-2.097941 < -2.094395).

liangfok, 2016-06-08:
@amcastro-tri: Do you know of a good way to enforce joint limits? I was thinking about exponentially increasing the joint-limit forces before a joint hits its limit such that it becomes infinity when the limit is actually hit, but his seems extremely risky from a simulation stability point of view.

amcastro-tri, 2016-06-08:
Liang. As we saw yesterday in `RigidBodySystem::dynamics` joint limits are modeled as a spring/damper  system. The spring stiffness is `penetration_stiffness` and the damper's damping is `penetration_damping`. It looks similar to a PD controller but physically is more like a spring/damper system. 
Therefore these limits are not enforced directly say, using a constraint together with an LCP solver, but by using this spring approximation (it is an approximation). Then there could be a violation of the joint limits but these should go to zero as the stiffness of these springs goes to zero. 
What I think it happens in your case is that the final pose of the arm is "under tension" against the ground  and the springs imposing the joint limits are not strong enough to move the arm against the friction with the ground. 

Yes, we would love to have stiffness and damping for collision to be a completely different thing from the springs used here. For collision, we were thinking with @sherm1 about a materials library and that each collision element would have its own material (similar to visuals and color). 
For the joint limits ideally I think these springs should be part of the joint itself, not a global quantity. 

With our engine as it is right now, you have to just know that those differences are modeling errors and that those modelling errors decrease with stiffness. 

I would say that for this case you make these comparisons to be within a tolerance (in contrast of these very precise comparisons). Something like `position < min_limit[ii]-tolerance` and `position > max_limit[ii] + tolerance`. The tolerance would need to be as big as 0.02 (1.15 degrees, not that bad).

liangfok, 2016-06-08:
@amcastro-tri, thanks for the explanation. I will for now do as you suggest and include a tolerance when checking joint tolerance. I look forward to comparing the spring-damper method against the constraint-based method.

amcastro-tri, 2016-06-08:
I do too!

closed 2016-06-08 by jwnimmer-tri