RobotLocomotion/drake #2517: Add support for joint velocity limits

liangfok, 2016-06-08:
First update the SDF and URDF parsers to parse and save these limits in `DrakeJoint`. Then update the simulation to enforce these limits.

sammy-tri, 2019-02-19:
Reassigning to @sherm1 to triage/assign within the dynamics team.  cc @amcastro-tri

sammy-tri, 2019-02-19:
(In case it wasn't clear, I removed myself because #10691 merged so the URDF/SDF parser part is done)

sherm1, 2019-02-19:
To actually _enforce_  joint velocity limits in a physical simulation requires a model of a physical mechanism for doing so. For example, the limits could be enforced by an active controller that generates velocity-opposing motor torques, or a passive mechanism like a nonlinear damper. A nonholonomic unilateral constraint could conceivably be used.

Is it reasonable to pick a mechanism and automatically add it to a robot simulation? My inclination would be just to allow people access to the specified velocity limits to use in their own models. If we do it automatically, what mechanism would robot folks like to see employed?

jwnimmer-tri, 2021-11-12:
@calderpg-tri and @hongkai-dai would you like to weigh in here, on whether there is any default enforcement mechanism that you would like to see added to Drake?  If we can't think of any, we'll close the issue.

calderpg-tri, 2021-11-12:
I'm fine with no enforcement of velocity limits by the simulator - I'm not sure there's a generic physical mechanism that would be suitable for modelling a velocity limit in the way we use them. I would venture that in most robots the physical mechanism is capable of motion at significantly higher speeds than the controller allows (and that would be present as velocity limits in a URDF/SDF) and that the controllers limit speed for more basic reasons (safety, stopping distance, max number of encoder ticks/sec without missing interrupts, etc). Not like position limits that represent physical limitations on joint motion.

Controllers can obey/enforce velocity and acceleration limits, and we should have examples in Drake of controllers that do so, but that's entirely separate from simulation behavior.

hongkai-dai, 2021-11-12:
Agree with @calderpg-tri . We currently use joint velocity limits in both the planner and the controller, and the velocity limit we use in the planner/controller is actually much smaller than the actual maximal velocity of the robot joints.

jwnimmer-tri, 2021-11-12:
Ok.  I just double-checked that we do parse joint limits for revolute and prismatic joints from sdformat into the MbP.  As such, I think we can call this issue finished.

closed 2021-11-12 by jwnimmer-tri