RobotLocomotion/drake #2518: problem with the installation from the source code

palmieri, 2016-06-08:
Hi all,

I am trying to install drake by compiling its source. I got the src from https://github.com/RobotLocomotion/drake.git.
I am running Ubuntu 14.04, with gcc/g++ 4.9.3 and cmake 3.5. The installation is carried on with the DISABLE_MATLAB option ON (because I would like to use only the cpp API).
After giving the command "make" from the main drake folder, I got this error:

> Performing configure step for 'bot_core_lcmtypes'
> -- CMAKE_INSTALL_PREFIX=/home/palmieri/libraries/drake/install
> CMake Error at cmake/pods.cmake:721 (message):
> 
> 
> 
> 
>         Do not run cmake directly in the pod directory.
>         use the supplied Makefile instead!  You now need to
>         remove CMakeCache.txt and the CMakeFiles directory.
> 
> 
> 
>         Then to build, simply type:
>          $ make
> 
> Call Stack (most recent call first):
> cmake/pods.cmake:761 (enforce_out_of_source)
> CMakeLists.txt:5 (include)

Thanks for your help!
Best,

Luigi

david-german-tri, 2016-06-08:
This is odd!  I was unable to reproduce just now in a clean clone of `master` at de9d89 on Ubuntu 14.04 with `g++ 4.9.3` and `cmake 3.5.2`.

```
git clone ...
make options # set DISABLE_MATLAB to OFF
make
```
One possible suspect: your `CMAKE_INSTALL_PREFIX` ends with `/install`, whereas mine ends with `/build`.  Any idea how that happened?

liangfok, 2016-06-08:
I've seen this error before when I accidentally ran:

```
$ cd [drake-distro]/drake/
$ cmake .
```

Is this what you did?

palmieri, 2016-06-08:
Yes, that's what I erroneously did (cmake . )! Thanks!

liangfok, 2016-06-08:
Great! I will go ahead and close this issue. Please feel free to re-open it if you're still having problems compiling Drake.

closed 2016-06-08 by liangfok