RobotLocomotion/drake #2535: Running tests and examples in Visual Studio requires PATH to be set

sherm1, 2016-06-09:
Normal workflow in Visual Studio is: change code, build affected libraries, run tests. The expectation is that the tests are using the libraries just built in the same VS solution. This doesn't work in the Drake build system unless PATH has been changed to include the binary directory for new libraries, and apparently also a related install location. That makes the fix proposed in #2094 awkward because switching between the Debug and Release build directories doesn't work unless you also remember to change the PATH environment variable. Failure to switch generally causes executables to fail (good). Worse is if the PATH is pointing to a different build directory that works, but doesn't reflect the code changes just made. I have wasted time that way assuming my code change didn't work.

Since Windows looks first in an executable's directory to find DLLs it depends on, one possibility would be to put new DLLs and executables in the same directory (DLLs normally go in `bin` anyway). These can be untangled by an `install` which should not be necessary during debug-change-build-test cycles.

(Moved discussion from end of #2094.)

sherm1, 2016-06-09:
From #2094:
> @mwoehlke-kitware said: Pedantically, the only way to enforce that is with RPATH, which doesn't exist on Windows. (Practically, on Windows — which doesn't use separate paths for binaries and libraries, and implicitly places . in %PATH% — the installed tests and corresponding libraries should be in the same folder. Generally speaking, .dll's should be installed to bin. Fixing that likely is sufficient for this purpose.)

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri