RobotLocomotion/drake #2537: Current implementation of coulomb friction (at the joint level) is incorrect

RussTedrake, 2016-06-10:


kuindersma, 2016-06-10:
We used this approximation to avoid chattering around the velocity zero-crossing when performing inverse dynamics on Atlas. Chapter 9 of Khalil and Dombre includes this model.

RussTedrake, 2016-06-10:
i see.  For that case, presumably we were never actually at zero measured velocity?  Do you think a more useful model would have some deadband for the sticking?

kuindersma, 2016-06-10:
I'm not sure I'm completely parsing the model you're proposing, but I think as long as there's velocity noise, any model with a discontinuous jump will potentially cause chattering problems.

kuindersma, 2016-06-10:
Oh, nvm! Yeah, I think what you're saying could work.

RussTedrake, 2017-02-19:
this is matlab-only, and is no longer relevant (the new models in c++ are correct).

closed 2017-02-19 by RussTedrake