RobotLocomotion/drake #2539: Change cpplint from a system dependency, to a superbuild external

jwnimmer-tri, 2016-06-10:
We should change cpplint from a system dependency, to a superbuild external.  Different versions will give different errors, and we should all be using the same version.  This may also allow us to tweak cpplint to our liking, if need be.

jamiesnape, 2016-06-15:
Can we just commit it to this repo? It is only one Python file.

jwnimmer-tri, 2016-06-15:
That's a good consideration.  Let me think about it.

jwnimmer-tri, 2016-07-12:
WIP https://github.com/jwnimmer-tri/drake/tree/cpplint-as-external

The `cpplint.py` is just one file, but it's ~250kB, so I'm exploring doing it as an external.  Also, it might end up being nice to have a copy of googlestyle IDE plug-ins in a well-known place for us to reference in our documentation.

jamiesnape, 2016-08-12:
Jenkins is now using cpplint from superbuild.

closed 2016-08-12 by jamiesnape

mwoehlke-kitware, 2016-08-12:
...and the cpplint build now seems to be broken: https://drake-jenkins.csail.mit.edu/job/experimental/3608/compiler=cpplint,label=linux,type=debug/ ...`-DWITH_DRAKE=OFF` missing?

jamiesnape, 2016-08-12:
@drake-jenkins-bot linux-cpplint-experimental please

jamiesnape, 2016-08-12:
The standalone job worked: https://drake-jenkins.csail.mit.edu/view/Lint/job/linux-cpplint-experimental/164/

mwoehlke-kitware, 2016-08-12:
@jamiesnape, on second thought... maybe it's not doing a merge build? (Are the default experimental builds not merge builds?)

jamiesnape, 2016-08-12:
No, they are merge, unless you have a merge conflict.

jamiesnape, 2016-08-12:
> ...unless you have a merge conflict.

Which you do in this instance.

mwoehlke-kitware, 2016-08-12:
> ...unless you have a merge conflict. Which you do in this instance.

Bother :weary:... okay, seems better now; thanks.